nodemcu/nodemcu-firmware #191: Api to stop consuming packets and drain event for socket

Point-less, 2015-02-08:
While trying to do a performnace test on nodemcu's tcp sockets i tried to built an echo server and sent it 1 megabyte of data. The naive implementation (each packet it that gets received gets inmediatly sent again) failed miserabily as packets came more fast than they could be sent and i end up receiving from 200 to 500 kb of data.
This implementation can be found in here https://gist.github.com/Point-less/5734639643bae814c259.

Am i missing something or is there a way of doing this with the current api? Can this be implemented? I don't know the limitations of the sdk, but if i get confirmation of the posibility of implementing and the need of this i may get my hands dirty with the code and make a PR.

Point-less, 2015-02-08:
Okay, by looking at espconn.h of sdk 0.9.5 i found the functions

/******************************************************************************
 * FunctionName : espconn_recv_hold
 * Description  : hold tcp receive
 * Parameters   : espconn -- espconn to hold
 * Returns      : none
*******************************************************************************/
sint8 espconn_recv_hold(struct espconn *pespconn);

/******************************************************************************
 * FunctionName : espconn_recv_unhold
 * Description  : unhold tcp receive
 * Parameters   : espconn -- espconn to unhold
 * Returns      : none
*******************************************************************************/
sint8 espconn_recv_unhold(struct espconn *pespconn);

So the throttling of the TCP should be doable from LUA.

By reading info on several sites, it seems that the current SDK does not allow more than 1400 bytes or so of buffered data, is there any workaround this? Using the "sent" callback is REALLY slow for packet-per-packet transmission (no TCP windowing).

Point-less, 2015-02-09:
Made a pool request adding the hold/unhold methods to sockets and implementing them in lwip

Point-less, 2015-02-09:
Okay, while i was bored i started to see how the second part could be implemented and by the time i realize i had half of the implementation made. I have a working modification which returns the amount of bytes succesfully dispatched to the stack on socket:send, so we know exactly how many bytes went out of the wifi. On the sockets's sent callback i've added a parameter which tells how many bytes where ACKed. This second hack i've made is a little messy, as there are some modifications to be done on the UDP API and i haven't done the SSL sockets. If there is interest in this i'll finish it up and do a PR.

lmamakos, 2015-02-11:
Wouldn't it make more sense for the callback to pass along the current size of the local send window?  (That is, how many bytes its safe to send because there is sufficient buffering available to hold it until is is ACKed by the remote TCP).    Done this way, you don't have to worry about the send call discarding any data you pass it.

Point-less, 2015-02-11:
The right thing would be to signal the error on the send call (now it's just sending nil) and if we signal an error we can just tell the sent bytes (been zero none).
Also, i've been thinking on how to do this on the SSL module, and i came to realize that the "free sent bytes" on the sent callback are useless as bytes of data sent are not the same as bytes of data on buffer (you can't know how much buffer are you going to use until you encript the data). In the same way, your proposition would suffer from the same problem, knowing the buffer available would not tell anything about how much data can be sent. 
I will be reverting my modification on the sent callback and do a PR with the modifications on the espconn_sent and espconn_secure_sent which returns the amount of bytes accepted by the stack.

Point-less, 2015-02-13:
Okay, i came to realize that the espresiff implementation already handled the case (i was reinventing th wheel and exposing it's internals to userspace) in which the data does not fit in the send buffer, the only thing that has to be solved is the nodemcu limit on 1460 bytes per sent (which can be raised to anything we want, as the TCP stack wont stop us from doing so)

marcelstoer, 2016-04-10:
> the only thing that has to be solved is the nodemcu limit on 1460 bytes per sent

I don't agree that this has to be solved -> closing. Fellow committers feel free to re-open if you disagree.

closed 2016-04-10 by marcelstoer

TerryE, 2016-04-11:
Marcel :+1:  This is as an SDK feature / constraint. We've had lots of discussions on this general topic: in general we shouldn't try fix SDK issues unless the fix is simple and the benefits are there.