nodemcu/nodemcu-firmware #193: Captive Portal

inversesandwich, 2015-02-08:
Would it be possible to have a captive portal function for when the ESP8266 is being used as an AP

raz123, 2015-02-09:
See: https://github.com/hamishcunningham/fishy-wifi/tree/master/joinme

inversesandwich, 2015-02-14:
@raz123 That is not a captive portal, I still have to browse to 192.168.4.1

slaff, 2015-02-16:
Such a functionality would be great! Is EPS8266 providing its own DNS server? May be a fake DNS server can be written that redirects all  hosts to 192.168.4.1 when the "captive" mode is activated.

layerzerolabs, 2015-02-26:
Hiya - fishy-wifi people here - we're still working on exactly that 'fake' dns server idea to achieve a true captive portal. If you'd like to collaborate to speed this feature up then pls ping us a message. Cheers!

marcelstoer, 2015-12-12:
See http://www.areresearch.net/2015/09/new-captiveintraweb-release-v3-esp8266.html.

closed 2015-12-12 by marcelstoer