nodemcu/nodemcu-firmware #202: net and tmr conflict

myplacedk, 2015-02-11:
I've been having issues with creating TCP connections, and I've finally narrowed it down to be some kind of conflict between net and tmr. I can recreate it reliably with this script:
```lua
print(wifi.sta.getip()) -- Just to confirm it's connected
tmr.delay(1000000)
conn=net.createConnection(net.TCP, false) 
conn:on("connection", function(conn)
    print("Connected")
    conn:send("GET / HTTP/1.1\r\nHost: www.nodemcu.com\r\n"
        .."Connection: keep-alive\r\nAccept: */*\r\n\r\n")
end)
conn:on("sent", function(conn)
    print("Sent data")
end)
conn:on("receive", function(conn, pl)
    print("Received data")
    conn:close()
end)
conn:on("disconnection", function(conn)
    print("Disconnected")
end)
print("Connecting...")
conn:connect(80,"121.41.33.127")
```

init.lua is empty and I run the script a few seconds after reset. I get this:

```
_ÇÏRSöâFjSöâFjúfî¡Cá
NodeMCU 0.9.5 build 20150127  powered by Lua 5.1.4
> dofile('test.lua')
192.168.0.103	255.255.255.0	192.168.0.1
Connecting...
> Connected
Disconnected
```

Sometimes I don't get "Connected". Sometimes I don't even get "Disconnected" until next time I try to open a connections.

If I comment out the delay, I get this (sometimes I have to power-cycle the ESP8266 before it works):
```
8Ÿ†þŠƒ(‡.,Š–Š‚‡(‡.,
‹ƒ‡£¡Cá
NodeMCU 0.9.5 build 20150127  powered by Lua 5.1.4
> dofile('test.lua')
192.168.0.103	255.255.255.0	192.168.0.1
Connecting...
> Connected
Sent data
Received data
Disconnected
```

I've tried several of the recent pre_build firmwares, and I had problems with all of them. However this exact example is only tested on the latest firmware.

nodemcu, 2015-02-11:
tmr.delay will stop everything. include wifi beacon.
so after tmr.delay for a long time. your wifi connection will be beacon time out.
you have to reconnect to ap.
tmr.delay are not recommended in long time(1 second) delay.
please use tmr.alarm instead.

myplacedk, 2015-02-11:
That makes sense. I already implemented tmr.alarm as a workaround. I'll add a comment in the wiki later, if nobody else does it first.

closed 2015-02-11 by nodemcu