nodemcu/nodemcu-firmware #210: loosing WiFi (station )connection

iMarvinS, 2015-02-12:
I'm using NodeMCU 0.9.5 build 20150212 on the ESP-07 and ESP-12.
Both are loosing WiFi connection to my accesspoint (WD MyNet N600) after some minutes (I can´t really measure it).
The connection only gets lost after some minutes of inactivity (no TCP socket is connected).
If I continuously ping the IP the connection does not get lost and I can send requests to it every time.
The @nodeMCU support on Twitter already told me to use a external antenna with the ESP-07 but it also does not work (also isn´t applicable on the ESP-12 ).

Does anyone have advice for this issue?

vowstar, 2015-02-13:
Is this ap?
[image]
[image]

I think WD MyNet N600 is special, it is may a SDK bug. I'll send this issue to espressif and they will test how to support this ap.

iMarvinS, 2015-02-13:
That´s the right one.
I´ll try it with a FRITZ!Box 7362 provided by my ISP.

vowstar, 2015-02-13:
This issue have replied by espressif. Some ap will kick station during sometime have no communication, it is normal.

iMarvinS, 2015-02-13:
This is indeed a problem with the N600 Accesspoint. With a FRITZ!Box both modules work just fine. 
After analyzing some logs of the N600 I saw the issue described by espressif.

Maybe a DD-WRT installation on the N600 can help to solve the issue.

For now this issue can be closed.

closed 2015-02-13 by iMarvinS