nodemcu/nodemcu-firmware #211: NodeMCU webserver cant serve big files (about 4Kb's)

aguaviva, 2015-02-13:
to repro use the firmware nnodemcu_20160126.bin upload the code below. Connect your browser to the esp8266. You will see your all your files listed. Click on the small ones, they show up in the browser fine. click on any file bigger than 3Kb.. you get it incomplete

I recommend giving a try to this script is really cool.

'''javascript
--[[
 XChip's NodeMCU IDE

 Edit NodeMCU files using your webbrowser.

]]--

rnrn=0
Status = 0
DataToGet = 0
method=""
url=""
vars=""

function Sendfile(conn, filename)
    if file.open(filename, "r") then
        repeat
            local line=file.read(128)
            if line then conn:send(line)end
        until not line    
        file.close()
    end
end

srv=net.createServer(net.TCP) 
srv:listen(80,function(conn) 
  conn:on("receive",function(conn,payload) 
    
    if Status==0 then
        _, _, method, url, vars = string.find(payload, "([A-Z]+) /([^?]*)%??(.*) HTTP")
        print(":",method, url, vars)                          
    end
    
    if method=="POST" then

        if Status==0 then
            --print("status", Status)
            _,_,DataToGet, payload = string.find(payload, "Content%-Length: (%d+)(.+)")
            if DataToGet~=nil then
                DataToGet = tonumber(DataToGet)
                --print(DataToGet)
                rnrn=1
                Status = 1                
            else
                print("Can't get length")
            end
        end
        
        -- find /r/n/r/n
        if Status==1 then
            --print("status", Status)
            local payloadlen = string.len(payload)
            local mark = "\r\n\r\n"
            local i
            for i=1, payloadlen do                
                if string.byte(mark, rnrn) == string.byte(payload, i) then
                    rnrn=rnrn+1
                    if rnrn==5 then 
                        payload = string.sub(payload, i+1,payloadlen)
                        file.open(url, "w")
                        file.close() 
                        Status=2
                        break
                    end
                else
                    rnrn=1
                end
            end    
            if Status==1 then 
                return 
            end
        end       
    
        if Status==2 then
            --print("status", Status)
            if payload~=nil then
                DataToGet=DataToGet-string.len(payload)
                --print("DataToGet:", DataToGet, "payload len:", string.len(payload))
                file.open(url, "a+")
                file.write(payload)            
                file.close() 
            else
                conn:send("HTTP/1.1 200 OK\r\n\r\nERROR")
                Status=0
            end

            if DataToGet==0 then                
                conn:send("HTTP/1.1 200 OK\r\n\r\nOK")
                 
                if vars=="run" then
                    conn:send("<br>Result:<verbatim>")                
                    local st, result=pcall(dofile, url)
                    conn:send(tostring(result))   
                    conn:send("</verbatim>")
                end
                
                Status=0
            end            
        end
        
        return
    end    
    
    if url == "favicon.ico" then
        conn:send("HTTP/1.1 404 file not found")
        return
    end

    conn:send("HTTP/1.1 200 OK\r\n\r\n<html><body>")
    conn:send("<h1>NodeMCU IDE</h1>")
    
    if url=="" then
        local l = file.list();
        for k,v in pairs(l) do  
            conn:send("<a href='"..k.."'>"..k.."</a>, size:"..v.."<br>")
        end
    else        
        conn:send("<script>function tag(t,c) {document.getElementsByTagName(t)[0].innerHTML=c} function upload(f,d){ tag('w','Saving'); var x = new XMLHttpRequest(); x.open('POST', '/'+f, true); x.onreadystatechange=function() { if (x.readyState==4) tag('w',x.responseText); }; x.send(new Blob([document.getElementsByName('t')[0].value], {type: 'text/plain'}));} </script>")
        conn:send("<a href='/'>Back to file list</a><br><br><textarea name=t cols=79 rows=17'>")
        Sendfile(conn, url)
        conn:send("</textarea></br>")   
        conn:send("<button onclick=\"upload('"..url.."')\">Save</button>")
        conn:send("<button onclick=\"upload('"..url.."?run')\">Save+Run</button><br><w></w>")
    end
    
    conn:send("</body></html>")    
    
  end)
  conn:on("sent",function(conn) conn:close() end)
end)
print("listening, free:", node.heap())
'''

Point-less, 2015-02-13:
https://github.com/nodemcu/nodemcu-firmware/issues/191

I'll be doing a PR today with the changes needed for this to work. Currently there is a limit of 3 1460 packets on the output buffer (ad wihtout a proper api you don't know how much buffer you have left)

nodemcu, 2015-02-13:
increase output buffer will not solve this issue. only increase the limit from 3k to say 6k.
you should rewrite the  
```lua
function Sendfile(conn, filename)
    if file.open(filename, "r") then
        repeat
            local line=file.read(128)
            if line then conn:send(line)end
        until not line    
        file.close()
    end
end
```

split the file into 1460s, send and after the event "sent" fired, send another one.

Point-less, 2015-02-13:
My proposition is not to increase the buffer, but manage it in a more efficient way:

```
s=net.createServer(net.TCP,180) 
s:listen(10000,function(c) 
	local datalen = 0
	local remainder = ""
	print("C")	
	
	local enviar = function(c, l)
		local to_send = remainder..l
		local amount_sent
		remainder = ""
		if string.len(to_send) > 0 then
			print("Sending "..string.len(to_send))
			amount_sent = c:send(to_send)
			print("Sent "..amount_sent)
			if amount_sent < string.len(to_send) then
				c:hold()
				remainder = string.sub(to_send, amount_sent)
				print("Saving "..string.len(remainder))
			end
		else
			amount_sent = 0
		end
		
		return amount_sent
	end
	
	c:on("sent",function(c)
		print("A")
		c:unhold()
		
		datalen = datalen + enviar(c, "")
	end)
	
    c:on("receive",function(c,l)
		print("R"..string.len(l))
		local amount_sent = enviar(c, l)
		
		datalen = datalen + amount_sent
    end)
	
    c:on("disconnection",function(c) 
		print("D "..datalen)
    end)
end)
```

An echo server would look like this. 
By knowing how many bytes you get on the stack you can use all the output buffer (currently 4380 bytes).

Point-less, 2015-02-13:
Disregard my comment, i won't push the modification, as it was already handled by the sdk, what we must address now is the 1460 hard limit which nodemcu has on sent's function data (which can only be  called once until the sent callback is called)

aguaviva, 2015-02-13:
Oh I see, then I have to split the data I want to send in several 1460 bytes chunks. The first one should be sent in the "receive" callback and the rest in the "sent". Thanks for this!

beckmx, 2015-02-13:
I tried with latest build from the 13 and it looks it crashes the chip, it keeps printing random things like NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4
>       "@   �� ��?8   ؏�?0       ��z�(�

any ideas?

dvv, 2015-02-13:
It 's all about lack of memory. When do we have offline unit tests?

nodemcu, 2015-02-13:
may it's the internal buffer size I raised cause some lack of memory.
I push a revert here in dev branch, and build a bin. too late in China to test.
https://github.com/nodemcu/nodemcu-firmware/blob/dev/pre_build/latest/nodemcu_latest.bin

nodemcu, 2015-02-13:
and call for unit tests. can any body write some python or even lua script to test the firmware through serial port :)

beckmx, 2015-02-13:
I will test it..

nodemcu, 2015-02-13:
thanks,
in ver 20150212 it works right?

beckmx, 2015-02-13:
okay, everything looks good, but still the problem of 2kb buffer size is still there, it still cuts the transmission when sending the contents of the file

aguaviva, 2015-02-13:
Ok I fixed the script according to your feedback, now it is working fine. I think the current way nodemcu works is the right one. If you don't use the callback method then each time the sending buffer is full you should have to block the script until the data is sent. 

  

```javascript
--[[
 XChip's NodeMCU IDE
 Edit NodeMCU files using your webbrowser.
Just note that the script cannot display itself because of the </textarea> tag. It still is very useful.
]]--

srv=net.createServer(net.TCP) 
srv:listen(80,function(conn) 

   local rnrn=0
   local Status = 0
   local DataToGet = 0
   local method=""
   local url=""
   local vars=""

  conn:on("receive",function(conn,payload) 
    
    if Status==0 then
        _, _, method, url, vars = string.find(payload, "([A-Z]+) /([^?]*)%??(.*) HTTP")
        print(method, url, vars)                          
    end
    
    if method=="POST" then
    
        if Status==0 then
            --print("status", Status)
            _,_,DataToGet, payload = string.find(payload, "Content%-Length: (%d+)(.+)")
            if DataToGet~=nil then
                DataToGet = tonumber(DataToGet)
                --print(DataToGet)
                rnrn=1
                Status = 1                
            else
                print("bad length")
            end
        end
        
        -- find /r/n/r/n
        if Status==1 then
            --print("status", Status)
            local payloadlen = string.len(payload)
            local mark = "\r\n\r\n"
            local i
            for i=1, payloadlen do                
                if string.byte(mark, rnrn) == string.byte(payload, i) then
                    rnrn=rnrn+1
                    if rnrn==5 then 
                        payload = string.sub(payload, i+1,payloadlen)
                        file.open(url, "w")
                        file.close() 
                        Status=2
                        break
                    end
                else
                    rnrn=1
                end
            end    
            if Status==1 then 
                return 
            end
        end       
    
        if Status==2 then
            --print("status", Status)
            if payload~=nil then
                DataToGet=DataToGet-string.len(payload)
                --print("DataToGet:", DataToGet, "payload len:", string.len(payload))
                file.open(url, "a+")
                file.write(payload)            
                file.close() 
            else
                conn:send("HTTP/1.1 200 OK\r\n\r\nERROR")
                Status=0
            end

            if DataToGet==0 then                
                conn:send("HTTP/1.1 200 OK\r\n\r\nOK")     
                Status=0
            end            
        end
        
        return
    end    
    
    DataToGet = -1
    
    if url == "favicon.ico" then
        conn:send("HTTP/1.1 404 file not found")
        return
    end

    conn:send("HTTP/1.1 200 OK\r\n\r\n<html><body><h1>NodeMCU IDE</h1>")
    if url=="" then
        local l = file.list();
        for k,v in pairs(l) do  
            conn:send("<a href='"..k.."'>"..k.."</a>, size:"..v.."<br>")
        end        
    end
    
    DataToGet = 0
    
  end)
  conn:on("sent",function(conn) 
  
    if DataToGet>=0 and method=="GET" then
    
        if vars=="run" then
        
            conn:send("<verbatim>")                
            local st, result=pcall(dofile, url)
            conn:send(tostring(result))   
            conn:send("</verbatim>")  
            
        elseif url~="" then
        
            if DataToGet==0 then 
                conn:send("<script>function tag(c){document.getElementsByTagName('w')[0].innerHTML=c} function upload(f,d){tag('Saving');var x =new XMLHttpRequest();x.open('POST','/'+f,true);x.onreadystatechange=function(){if(x.readyState==4) tag(x.responseText);};x.send(new Blob([document.getElementsByName('t')[0].value],{type:'text/plain'}));}</script>")
                conn:send("<a href='/'>Back to file list</a><br><br><textarea name=t cols=79 rows=17>")
            end
        
            if file.open(url, "r") then            
                file.seek("set", DataToGet)
                local line=file.read(512)
                file.close()
                if line then
                    conn:send(line)
                    DataToGet = DataToGet + 512    
                
                    if (string.len(line)==512) then
                        return
                    end
                end
            end
        
            conn:send("</textarea></br>")   
            conn:send("<button onclick=\"upload('"..url.."')\">Save</button>")
            conn:send("<a href='/"..url.."?run'>run</a><w></w>")
        end

        conn:send("</body></html>")
    end
    
    conn:close() 
  end)
end)
print("listening, free:", node.heap())
```

beckmx, 2015-02-14:
I put the above code in the init.lua and it says:

NodeMCU 0.9.5 build 20150214  powered by Lua 5.1.4
lua: not enough memory
>

nodemcu, 2015-02-14:
I am testing this ok
```lua
NodeMCU 0.9.5 build 20150214  powered by Lua 5.1.4
listening, free:        13200
>
```
could you
```lua
print(node.heap())
```
after your flashing the firmware?
and do file.format() to clear file system

aguaviva, 2015-02-14:
Please feel free to add my script to the repo if you find it useful. It should be called webide.lua

GeoNomad, 2015-02-14:
> Is this script working well for you guys? Any bugs/issues?

Works great. Very nice job aquaviva.

I would insert `wifi.setmode(wifi.SOFTAP)` at the beginning as it can't be assumed.

beckmx, 2015-02-14:
what branch are you using dev or master?, it is still pulling the bug

GeoNomad, 2015-02-14:
20150212 prebuild

beckmx, 2015-02-14:
well I changed the firmware to  day 13th and it keeps resetting
NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4
listening, free:	11720

repetead all the time, I will try with day 12

reischle, 2015-03-25:
The trouble was in the increased buffer size of the 20150213 build. I also had problems with that. Aguaviva's code is superb! I'll probably build one of my projects on that basis.

creationix, 2015-05-28:
I just implemented websocket for nodemcu.  All you need is my new crypto module https://github.com/nodemcu/nodemcu-firmware/pull/447 and my POC server https://gist.github.com/creationix/079e11345249525c7e2e.

Currently my code doesn't handle large chunks, but the websocket deframer does support message frames being split across multiple data events and/or network packets.

I could see this helping with some of the commands in the neat little IDE.

pastukhov, 2015-05-29:
@creationix 
https://github.com/marcoskirsch/nodemcu-httpserver/ 
The webserver, that can handle any size files. 
Could you help to integrate websockets into that project?

creationix, 2015-05-29:
@pastukhov neat webserver, but how would you use websockets with it?  My use case is a web-based IDE for developing on the nodemcu and I use the websocket channel as a binary-safe socket for the rpc commands.

Also I still need to finish my version to support loading and saving large files by using chunks.  I can see about integrating it with that server once I work out the major bugs in my version.

beckmx, 2015-05-29:
The large files might need to be loaded in parts because the ram in the
chip is not enough to store large files, not even to serve files while
processing other stuff, so, I could say we are trapped

pastukhov, 2015-06-01:
@creationix I'm planing to use websockets to implement web controlled devices with powerful js client side part.

saitejalakkimsetty, 2015-06-24:
@aguaviva I have modified your code to fit my needs. But it fails to serve the file.
```
wifi.setmode(wifi.STATION)
wifi.sta.config("myssid","mypassword")
srv=net.createServer(net.TCP)
srv:listen(80,function(conn)
	conn:on("receive",function(conn,payload) 
		local method=""
		local url=""
		local vars=""
		_, _, method, url, vars = string.find(payload, "([A-Z]+) /([^?]*)%??(.*) HTTP")
		print(method, url, vars)
		print("Printing the first")
		dofile("printer.lc")
		print("Printing the second")
		dofile("printer.lc")
		print("Printing the third")
		dofile("printer.lc")
		collectgarbage()
		print("Printing the fourth")
		dofile("printer.lc")
		print("Printing the fifth")
		dofile("printer.lc")
		print("Printing the sixth")
		dofile("printer.lc")
		collectgarbage()
		print("Printing the seventh")
		dofile("printer.lc")
		print("Printing the eighth")
		dofile("printer.lc")
		sender(url)
	end)
	
	function sender(uri)
		local DataToGet = 0
		conn:send("HTTP/1.1 200 OK\r\n\r\n")
		conn:on("sent",function(conn) 
			if file.open(uri, "r") then            
				file.seek("set", DataToGet)
				local line=file.read(1024)
				file.close()
				if line then
					conn:send(line)
					DataToGet = DataToGet + 1024
					if (string.len(line)==1024) then
						return
					end
				end
			end
			conn:close()
			collectgarbage()
		end)
	end
end)
print("listening, free:", node.heap())
```
The printer.lc has the following code:
```
for i=1, 100 do
	print(i)
end
```

What I am trying to do is serve a file that needs to be created when requested. It is a resource intensive task. To represent that, I am using the `printer.lc` But I cannot wrap my heap around why it fails. 

Thanks for taking a look.

aguaviva, 2015-06-24:
What is the expected output and what are you getting instead?

Note that print will just  print data to the serial port and it wont send those strings over the socket

saitejalakkimsetty, 2015-06-24:
The expected output is toe send the file who requested it. But is does not. But if i remove all the printer.lc statements in the "receive" part, it serves as expected.

Am I missing something?

aguaviva, 2015-06-25:
you should try removing bits until it works, that way you can isolate the offending bit and then ask again. The test case you are reporting is still too complex.

closed 2015-11-02 by TerryE