nodemcu/nodemcu-firmware #214: yet-another-dht22.lua: No tempereature, no humidity

FourQuarks, 2015-02-13:
With nodeMCU version 201510212 all works well.
With nodeMCU version 201510213 only these two errors:

DHT22   nil     0
or 
DHT22   nil     2

dvv, 2015-02-13:
May be caused by non-precise timing was introduced? https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_examples/yet-another-dht22.lua#L50
Try to play with wait value -- 20..40

FourQuarks, 2015-02-13:
Thank you for your tip. Alas, it didn't  work.

I don't believe it's the timing. There must be something else wrong with version 20150213.
As I wrote in version 20150212 the DHT22 program runs perfect.

dvv, 2015-02-13:
Then we need to look at the whole diff between versions, or even bisect the bug.

nodemcu, 2015-02-13:
may it's the internal buffer size I raised cause some lack of memory.
I push a revert here in dev branch, and build a bin. too late in China to test.
https://github.com/nodemcu/nodemcu-firmware/blob/dev/pre_build/latest/nodemcu_latest.bin

dvv, 2015-02-13:
Sleep well friend. Youreawesome

javieryanez, 2015-02-13:
Need a update to work with floating point. I need to update dht22.lua, and now work with integer or with floating point, like Manawyrm  with ds18b20.lua.

FourQuarks, 2015-02-14:
No change in behaviour in version 20150214 from the dev branch.

Without compilation and using
print(require("d22").read(4)) 
the error is 0, 3

With compilation, removing the LUA file and using
print(dofile("d22.lc").read(4))
the error is 0, 1

nodemcu, 2015-02-14:
can you confirm that 20150212 works?
in my test, the module is loaded correctly, but I have no dht22, so I got nil,0
in both 20150212 and 20150213

FourQuarks, 2015-02-14:
"yet-another-dht22.lua" works in 20150212

javieryanez, 2015-02-14:
Are you sure "yet-another-dht22.lua" works in 20150212? I see some things incompatible with floating point. For example, in line 63: ```if crc ~= b[5] then return nil, 3 end``` always true.

dvv, 2015-02-14:
Well, yet-another-*.lua use no floats at all, they rely on standard lua
features, so I see (so far?) no flaws in their code.

Hence, if their behaviors depend on whether the firmware support floats or
not, I believe it's a firmware flaw.

FourQuarks, 2015-02-15:
@javieryanez 

"For example, in line 63: if crc ~= b[5] then return nil, 3 end always true."
This is not correct as the code (I replaced return with print)  below shows.

" I see some things incompatible with floating point."
Which other things do you see?

NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4

> b   = {0, 1, 2, 3, 100}
> crc = 100
> if crc ~= b[5] then print (nil, 3) end     -- condition is false, no output
> crc = 101
> if crc ~= b[5] then print (nil, 3) end      -- condition is true, output nil     3
nil     3

javieryanez, 2015-02-15:
In my opinion, "yet-another-dht22.lua" no errors using integer variables. I do not see errors in the firmware.

Keep in mind two things:

1. "yet-another-dht22.lua" (and any other code) uses floating numbers if you run into a floating point firmware.

2. The other thing is that the comparison of floating numbers is different from integers.

It is easy to see problems comparing floating point numbers:

```
NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4
print(0.07*300)
21
print(0.07*300==21)
false
```
The consequence is that code programmed to integer variables does not always work when variables are floating point.

I changed dht22.lua code to work with integer variables and floating point. For example, to check the CRC:
```
if (checksumTest - checksum >= 1) or (checksum - checksumTest >= 1) then
    humidity = nil
  end
```

FourQuarks, 2015-02-15:
Your answer is misleading and without meaning and has nothing to do with my contribution.
You also didn't answer my question.

I wrote it  before: "yet-another-dht22.lua"  did run perfectly in earlier versions with floats.

closed 2015-02-15 by FourQuarks

javieryanez, 2015-02-15:
Misleading and without meaning??? I see very clearly that does not work "yet-another-dht22.lua". It is for the same reasons as dht22.lua not work before they changed. With the explanations, I have already given, takes very little to make it work. I could make a patch to make it work, but I have dedicated my time to make dht22.lua work.

I do not know why you think "misleading and without meaning" my explanation. I do not know what kind of response expected. Perhaps you expect a patch directly?.

dvv, 2015-02-15:
Let us cure the cause not conseqiences. There is no single division in
yetanother hence all the values must be exact. Hence the equality must be
strong.
Btw it forks for me with dth22 and stock peebuilt 20150212