nodemcu/nodemcu-firmware #217: Setting static IP in SOFTAP mode

jm01alternet, 2015-02-13:
I seem to have an issue with setting static IP in SOFTAP mode. 

I tried different approaches but all without success as they all return 192.168.4.1 
Although I do have great success when doing the same in sta mode :

    wifi.setmode(wifi.STATION)
    cfg={}
    cfg.ip="192.168.2.1"
    cfg.netmask="255.255.255.0"
    cfg.gateway="192.168.2.1"

    wifi.sta.setip(cfg)
    print(wifi.sta.getip())

I get :
192.168.2.1    255.255.255.0    192.168.2.1

When I try :

    wifi.setmode(wifi.SOFTAP)
    cfg={}
    cfg.ip="192.168.1.2"
    cfg.netmask="255.255.255.0"
    cfg.gateway="192.168.1.2"

    wifi.ap.setip(cfg)
    print(wifi.ap.getip())

I get :
192.168.4.1    255.255.255.0    192.168.4.1

Same applies in STATIONAP mode, the sta IP sets fine, but not the ap IP.

Seems like the ap.setip function does not "execute" ?

nodemcu, 2015-02-13:
which version of firmware do you use.
please ref #199  and #209

jm01alternet, 2015-02-13:
my fault.. didn't check it in the new version.. 

great work guys... thank you

closed 2015-02-13 by jm01alternet