nodemcu/nodemcu-firmware #218: require() not returning table in latest pre-build

Quantalume, 2015-02-13:
Works in nodemcu_20150212.bin but broken in nodemcu_latest.bin (20150213?).

    > dht22 = require("dht22")
    > dht22.read(PIN)
    stdin:1: attempt to index global 'dht22' (a boolean value)
    > print(dht22)
    true
    > 

More info here: http://www.esp8266.com/viewtopic.php?f=18&t=1620

nodemcu, 2015-02-14:
maybe same issue #214
try
https://github.com/nodemcu/nodemcu-firmware/blob/dev/pre_build/latest/nodemcu_latest.bin

nodemcu, 2015-02-14:
```lua
NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4
lua: cannot open init.lua
>
>
> dht22 = require("dht22")
> dht22.read(PIN)
dht22.lua:34: bad argument #1 to 'mode' (number expected, got nil)
> dht22.read(1)
> print(dht22)
table: 0x3fff5c50
```
can not re-produce this issue.
can you confirm? or anybody else can?

Quantalume, 2015-02-14:
I tried the 20150214 build, and it worked fine.  Then I reverted to the 20150213 build, and it is still working fine.  Perhaps there was some issue with the state of flash filesystem when I upgraded to 20150213 the first time.  Closing but will let you know if the issue resurfaces.  Thank you.

closed 2015-02-14 by Quantalume