nodemcu/nodemcu-firmware #219: Allow access to the RTC memory

slaff, 2015-02-13:
Allow LUA access to the system_rtc_mem_read and system_rtc_mem_write functions.

jrahlf, 2015-02-14:
How many bytes can we store?

slaff, 2015-02-15:
> @jrahlf : How many bytes can we store?
I am not quite sure. And definitely a limit is missing in the read and write functions. It would be great is someone can find in the scarce documentation.

GeoNomad, 2015-02-16:
It seems to be 512 bytes of user memory after the 256 bytes of system memory.

http://bbs.espressif.com/viewtopic.php?f=7&t=68&p=303&hilit=LIGHT_SLEEP_T#p292

During deep sleep, only RTC still working, so maybe we need to save some user data in RTC memory. Only “user data” area can be used by user. 
|_ _ _ _ _system data _ _ _ _ _|_ _ _ _ _ _ _ _ _ user data _ _ _ _ _ _ _ _ _|
| 256 bytes	| 512 bytes	|

Note: RTC memory is 4 bytes aligned for read and write operations. Parameter “des_addr” means block number(4 bytes per block). So, if we want to save some data at the beginning of user data area, “des_addr” will be 256/4 = 64, “save_size” will be data length.
Prototype：
bool system_rtc_mem_write (uint32 des_addr, void * src_addr, uint32 save_size)
Parameter：
uint32 des_addr —— destination address (block number) in RTC memory, 
des_addr >=64
void * src_addr —— data pointer.
uint32 save_size —— data length ( byte)
Return：
True， succeed；False，fail.

devsaurus, 2015-09-05:
Closed since request is implemented. Feel free to reopen in case the issue persists.

closed 2015-09-05 by devsaurus