nodemcu/nodemcu-firmware #223: udp and getpeer

hreintke, 2015-02-14:
Used the dev branch to compile a version with getpeer() option.
Works OK for tcp connections but get error  
"PANIC: unprotected error in call to Lua API (servertest1.lua:2: attempt to call method 'getpeer' (a nil value))"
when using it on an UDP server connection.

hreintke, 2015-02-15:
Have been investigating further on. 
This is the lua file I am using  : 
function pr(c,pl)
  print ("parameter info c")
  print(type(c), " ",getmetatable(c));
  print ("metadatainfo c");  
  mt = getmetatable(c);
  for n,s in pairs(mt) do print(n," ",s)  end
end

if srv ~= nil then srv:close() end
if udpsrv ~= nil then udpsrv:close() end

counter=1

srv=net.createServer(net.TCP)

srv:listen(80,function(conn) 
                conn:on("receive",pr)   
                conn:on("sent",function(conn) print("tcp package sent, Heap = "..node.heap()) conn:close() end)
               end)

udpsrv=net.createServer(net.UDP)
udpsrv:on("receive",pr)
udpsrv:on("sent", function(conn) print("udp packet sent \n") end)
udpsrv:listen(8000)

which results for UDP in 
parameter info c
userdata	 	romtable: 0x4023d100
metadatainfo c
listen	 	lightfunction: 0x4025d98c
close	 	lightfunction: 0x4025d03c
on	 	lightfunction: 0x4025ce04
send	 	lightfunction: 0x4025cb04
__gc	 	lightfunction: 0x4025c52c
__index	 	romtable: 0x4023d100

And for TCP in : 
parameter info c
userdata	 	romtable: 0x4023cff8
metadatainfo c
connect	 	lightfunction: 0x4025d974
close	 	lightfunction: 0x4025d024
on	 	lightfunction: 0x4025cdec
send	 	lightfunction: 0x4025caec
hold	 	lightfunction: 0x4025da10
unhold	 	lightfunction: 0x4025d9a4
dns	 	lightfunction: 0x4025dbf4
getpeer	 	lightfunction: 0x4025c54c
__gc	 	lightfunction: 0x4025c514
__index	 	romtable: 0x4023cff8

Still trying to find why there is this difference. 
Herman

hreintke, 2015-02-21:
LS,
I have not enough experience with Pull Request to get that done but the updates in this comment lead to a working getpeer() for servers (UDP).
1: add getpeer to the server metatable
```lua
static const LUA_REG_TYPE net_server_map[] =
{
  { LSTRKEY( "listen" ), LFUNCVAL ( net_server_listen ) },
  { LSTRKEY( "close" ), LFUNCVAL ( net_server_close ) },
  { LSTRKEY( "on" ), LFUNCVAL ( net_udpserver_on ) },
  { LSTRKEY( "send" ), LFUNCVAL ( net_udpserver_send ) },
  // { LSTRKEY( "delete" ), LFUNCVAL ( net_server_delete ) },
  { LSTRKEY( "getpeer" ), LFUNCVAL ( net_socket_getpeer ) },
  { LSTRKEY( "__gc" ), LFUNCVAL ( net_server_delete ) },
#if LUA_OPTIMIZE_MEMORY > 0
  { LSTRKEY( "__index" ), LROVAL ( net_server_map ) },
#endif
  { LNILKEY, LNILVAL }
};
```

2 : Update getpeer to : 
```lua
static int net_socket_getpeer( lua_State* L )
{
  NODE_DBG("Entered getpeer \n");
  lnet_userdata *nud;

  nud = (lnet_userdata *)luaL_testudata(L, 1, "net.server");
  if (nud == NULL)
  {
	  nud = (lnet_userdata *)luaL_testudata(L, 1, "net.socket");
  }
  luaL_argcheck(L, nud, 1, "Server/Socket expected");

  if(nud!=NULL && nud->pesp_conn!=NULL ){
      char temp[20] = {0};
      int tempPort = 0;
      c_sprintf(temp, IPSTR, IP2STR((nud->pesp_conn->type == TCP) ? &(nud->pesp_conn->proto.tcp->remote_ip) : &(nud->pesp_conn->proto.udp->remote_ip) ) );
      tempPort = (nud->pesp_conn->type == TCP) ? nud->pesp_conn->proto.tcp->remote_port :  nud->pesp_conn->proto.udp->remote_port;
      if ( tempPort != 0 ) {
        lua_pushstring( L, temp );
        lua_pushinteger( L, tempPort );
      } else {
        lua_pushnil( L );
        lua_pushnil( L );
      }
  } else {
      lua_pushnil( L );
      lua_pushnil( L );
  }
  return 2;
}
```
3: As I needed to check userdata to both net.server and net.socket, I cannot use the luaL_checkudata function. I exits with error instead of "just returning NULL" when not found.
Searching internet, I found in lua 5.1 no direct solution but lua 5.2.3 includes a new function luaL_testudata. I added that to net.c. Meaning when upgrading lua this should be removed from net.c.
```lua
LUALIB_API void *luaL_testudata (lua_State *L, int ud, const char *tname) {
  void *p = lua_touserdata(L, ud);
  if (p != NULL) {  /* value is a userdata? */
    if (lua_getmetatable(L, ud)) {  /* does it have a metatable? */
      luaL_getmetatable(L, tname);  /* get correct metatable */
      if (!lua_rawequal(L, -1, -2))  /* not the same? */
        p = NULL;  /* value is a userdata with wrong metatable */
      lua_pop(L, 2);  /* remove both metatables */
      return p;
    }
  }
  return NULL;  /* value is not a userdata with a metatable */
}
```
If you need more information, let me know.
Herman

TerryE, 2015-11-04:
SDK 1.4 Release notes Item 24:

> Modify the espconn pointer in the receive callback of UDP. Parameters remote_ip and remote_port in it are the remote IP and port set by espconn_create. If users want to obtain IP and ports of the current sender, please call espconn_get_connection_info to get relevant information. 

So the SDK now facilitates this and so we could consider this enhancement for the 1.4-based dev

lazy21tr, 2015-12-05:
As this is my first message on this project, thank you to all creaters/contributors of this project, nodemcu is a great firmware for esp modules.

@hreintke, I tried your patch and got "nil, nil" for getpeer() function. After some investigation, I found that both "temp" and "tempPort" are assigned 0 (in "net_socket_getpeer" function). I could not go further.

My environment: I use esp8266-01 and test my udp server code accross android phone, running "UDP Sender/Receiver" app from Google Play.

I need the peer IP to let the esp module reply to clients. Right now, I am planning to workaround this problem by appending client IP to its message and get it parsed in esp code so it can use this IP to create udp connection reply back to the sender.

However, I still want getpeer() function to work. Any help would be great.

hreintke, 2015-12-05:
@lazy21tr : 
It was a long time ago when I submitted the issue and currently am I not using nodemcu anymore for creating esp8266 firmware.

As you can see in the above comment, the project took the SDK 1.4.0 functionality to solve the issue.
You can check #750 and report any issues there.

sekarpdkt, 2016-04-08:
Is this issue resolved. Still getting same error attempt to call method 'getpeer' (a nil value))
Code:
`client=net.createConnection(net.UDP, 0)

port=8972
print("IP:"..wifi.sta.getip()..", Port:"..port)
srv=net.createServer(net.UDP)
srv:on("receive", function(srv, pl)
   -- rIP='192.168.42.1'
   rIP,rPort=srv:getpeer()
   print("From "..rIP..", reveived:"..pl)
    client:connect(8972,rIP)
    client:send("1234")
    client:close()
    client = nil
   end
)
srv:listen(port)`

I tried to use both dev and master branch.

thanks

TerryE, 2016-04-08:
OK, I will have a look at this and update this comment accordingly in 5 hours or so.

TerryE, 2016-04-08:
OK, I've tracked this one down.  It is related to #1080.  The underlying issue is that the UDP implementation is a bit of a botch.  A `net.createServer()` returns a `net.server` object which has methods `listen`,`on`,`send` and `close`, though `send()` and `on("sent")` only work in the case of UDP.  

A `net:server:on()` call passes a `net:server` object to the called function, but the `getpeer()` method is associated with a `net.socket` object and this is only created for TCP connections.  Catch-22.

As a quick workaround, I've just added the remote and local ip and ports as arguments to the receive callback and the remote IP:port is always reported as 0.0.0.0:0 in he espconn structure so I need to look at this further..

sekarpdkt, 2016-04-08:
Hi... Thanks. Will `srv:send("1234") ` work in case of UDP? Anyway I will try. Thanks.

TerryE, 2016-04-08:
Yes and it will send back to the same port as you are listening on.

TerryE, 2016-04-08:
OK, as well as this, there's a bug in , so the patch that I've added is
```diff
--- a/app/modules/net.c
+++ b/app/modules/net.c
@@ -145,6 +145,7 @@ static void net_socket_received(void *arg, char *pdata, unsigned short len)
 {
   NODE_DBG("net_socket_received is called.\n");
   struct espconn *pesp_conn = arg;
+  int nargs = 2;
   if(pesp_conn == NULL)
     return;
   lnet_userdata *nud = (lnet_userdata *)pesp_conn->reverse;
@@ -161,8 +162,18 @@ static void net_socket_received(void *arg, char *pdata, unsigned short len)
   // NODE_DBG(pdata);
   // NODE_DBG("\n");
   lua_pushlstring(gL, pdata, len);
-  // lua_pushinteger(gL, len);
-  lua_call(gL, 2, 0);
+  if (pesp_conn->type == ESPCONN_UDP) {
+    esp_udp *u = pesp_conn->proto.udp;
+    char ip_str[20] = {0};
+    c_sprintf(ip_str, IPSTR, IP2STR(&(u->remote_ip)));
+    lua_pushstring(gL, ip_str);   // the ip para
+    lua_pushinteger(gL, u->remote_port);
+    c_sprintf(ip_str, IPSTR, IP2STR(&(u->local_ip)));
+    lua_pushstring(gL, ip_str);   // the ip para
+    lua_pushinteger(gL, u->local_port);
+    nargs+=4;
+  }
+  lua_call(gL, nargs, 0);
 }
 
 static void net_socket_sent(void *arg)

--- a/app/lwip/app/espconn_udp.c
+++ b/app/lwip/app/espconn_udp.c
@@ -308,6 +308,11 @@ espconn_udp_recv(void *arg, struct udp_pcb *upcb, struct pbuf *p,
                wifi_get_ip_info(0, &ipconfig);
        }
 
+       precv->pespconn->proto.udp->remote_ip[0] = ip4_addr1_16(addr);
+       precv->pespconn->proto.udp->remote_ip[1] = ip4_addr2_16(addr);
+       precv->pespconn->proto.udp->remote_ip[2] = ip4_addr3_16(addr);
+       precv->pespconn->proto.udp->remote_ip[3] = ip4_addr4_16(addr);
+       precv->pespconn->proto.udp->remote_port = port;
        precv->pespconn->proto.udp->local_ip[0] = ip4_addr1_16(&ipconfig.ip);
        precv->pespconn->proto.udp->local_ip[1] = ip4_addr2_16(&ipconfig.ip);
        precv->pespconn->proto.udp->local_ip[2] = ip4_addr3_16(&ipconfig.ip);
```

and now a `srv:on("receive", function(sck, srcIP, srcPort, lclIP, lclPort) -- ...` works as expected. @jmattsson @pjsg, is this worth patching before #1080?

sekarpdkt, 2016-04-08:
Yes.. `srv:send("1234")`  worked.. I was not aware that we can listen on UDP client port also...
Node MCU Code: 
`
client=net.createConnection(net.UDP, 0)
port=8972
print("IP:"..wifi.sta.getip()..", Port:"..port)
srv=net.createServer(net.UDP)
srv:on("receive", function(srv, pl)
      print("received :"..pl)
      srv:send("1234") 
   end
)
srv:listen(port)
`

Node JS code:
`
var PORT = 8972;

var dgram = require('dgram');
var server = dgram.createSocket('udp4');

server.on('listening', function () {
    var address = server.address();
    console.log('UDP Server listening on ' + address.address + ":" + address.port);
});

server.on('message', function (message, remote) {
    console.log(remote.address + ':' + remote.port +' - ' + message);

});

server.bind(PORT);

var HOST = '192.168.42.239';

var message = new Buffer('Are you there?');

server.send(message, 0, message.length, PORT, HOST, function(err, bytes) {
    if (err) throw err;
    console.log('UDP message sent to ' + HOST +':'+ PORT);
});

`

TerryE, 2016-04-10:
I have decided not to add the extra parameters to the receive callback but to implement the `net:server:getpeer()` as the above commenters expect this to work.  Since #1233 is now addressing this, I am closing this issue.

closed 2016-04-10 by TerryE