nodemcu/nodemcu-firmware #225: Problems with DHT22 library and 2015-02-13 binary

robcazzaro, 2015-02-15:
I was using 2015-01-26 binary (prebuilt binary, didn't recompile anything) and started using the DHT22 library, which works very reliably. 

Updated to 2015-02-13, and updated the library to use floating point. But I started noticing a high rate of checksum errors (roughly 30-40% of the reads result in a checksum error). Went back to 2015-01-26 and, with the same library (the new one, compatible with both floating point and integers) and the exact same code, I can read the DHT22 very reliably, not a single checksum error in 50 tries

It looks as if the timing of the loops reading the DHT22 signal is slightly off, somehow. I know it's not a float math error, because when the checksum is off, either the temperature or humidity reading (usually humidity) is off significantly (i.e. while reading 45% humidity with the right checksum, might read 20% humidity when the checksum is off). I did put extra print() statements in the DHT22 library to see the values being read, btw, and not relying on the values returned by the function (which sets humidity to nil when the checksum is off)

robcazzaro, 2015-02-16:
I spent a bit more time understanding the DHT22 library, and I figured out how to solve the issue

This code reads the data, measuring the length of the pulse.

  for j = 1, 40, 1 do
    while (gpio_read(pin) == 1 and bitlength < 10 ) do
      bitlength = bitlength + 1
    end
    bitStream[j] = bitlength
    bitlength = 0
    -- bus will always let up eventually, don't bother with timeout
    while (gpio_read(pin) == 0) do end
  end

Short pulse=low, long pulse=high (from the DHT22 dayasheet)

In 2015-01-26, the values of bitStream[j] are between 0 and 2 for low (usually 1 and 2, 0 is only when the first bit is low), and 6 and 7 for high. For low, you see many 2, never a 5 for high

In newer builds (starting from 2015-01-27 up to and including 2015-02-13), the values of bitStream[j]  are 0-2 for low (here as well usually 1 and 2), but 4 and 5 for high. Very few 2 for low, and never a 6 for high. Very few 4 as well (that's why it rarely failed), most of the high seems 5, pointing to a slight timing difference in the loop

I guess that the "for j = 1, 40, 1 do" loop is now using floats, and much slower to execute. So the length of the pulse, measured by a slower loop, is inevitably shorter

My problem can be fixed by changing the following three lines in DHT22.lua:

    if (bitStream[i] > 4) then

    if (bitStream[i + 16] > 4) then

    if (bitStream[i + 32] > 4) then

to

    if (bitStream[i] > 3) then

    if (bitStream[i + 16] > 3) then

    if (bitStream[i + 32] > 3) then

(please note that the three lines are not contiguous)

As I thought, it's indeed a timing issue, that resulted in sometimes reading a 0 when should have been a 1 (for temperature, humidity or checksum), and resulting in problems

Using floats for a simple loop is overkill on such a small processor. But Lua has no way to force the use of integers, so the slower loop is here to stay, I guess (and until the DHT22 is implemented in C++ and offered as a built in API, I guess the DHT22 library must be changed)

In alternative, could you offer a pre-built bin using integers only? I know I could set it up and recompile myself, but would be nice not having to install the toolchain, since I really only would like to focus on the nodeMCU/Lua part

robcazzaro, 2015-02-19:
Looks like suggested change to dht22.lua was merged, Closing the issue

I have been using the new code for a couple of days, no problems at all

closed 2015-02-19 by robcazzaro

dvv, 2015-02-19:
Wonder if instead someone wants to port, say,
https://github.com/fasmide/esp_dht22/blob/master/user/user_main.c#L90, to
nodemcu. Being a C module, dth22 could gain more precise timing, hence
improved reliability.