nodemcu/nodemcu-firmware #230: UDP Server not working after close() and new createServer

hreintke, 2015-02-17:
LS,
I have the following code in the file udp_test.lua
```lua
if udpsrv ~= nil then udpsrv:close() end

counter = 1

function pr(c,pl)
  print(type(c), " ",getmetatable(c),"\n");
  print(type(pl), " ",getmetatable(pl),"\n");
  
-- remip, remport = c:getpeer();
-- print(remip,"\n",remport)
  outmsg=nil
  print (pl)
  outmsg = "Reply from esp, count = "..tostring(counter);
  c:send(outmsg)
end  

udpsrv=net.createServer(net.UDP)
udpsrv:on("receive",pr)
udpsrv:on("sent", function(conn) print("udp packet sent completed \n") end)
udpsrv:listen(8000)
```
When doing dofile("udp_test.lua") I do have a working udp server.
When doing a second time the dofile(udp_test.lua)
- I get no errormessages
- The UDP Server does not receive/reply to any udp message.
Only way to get udpserver running again is to restart the esp.

I added some additional debug code to net_socket_received. From that output I found that that call is returning due to "nud = NULL" and as such not forwarding the received data to the lua program.
```c
static void net_socket_received(void *arg, char *pdata, unsigned short len)
{
  NODE_DBG("net_socket_received is called.\n");
  struct espconn *pesp_conn = arg;
  if(pesp_conn == NULL) { NODE_DBG("pesp_conn Null \n"); return; }

  lnet_userdata *nud = (lnet_userdata *)pesp_conn->reverse;
  if(nud == NULL) { NODE_DBG("nud NULL \n"); return;}
```
This is the logfile from the test
```
dhcp client start...
ip:10.0.0.211,mask:255.255.255.0,gw:10.0.0.138
dofile('udp_test.lua')
net_create is called.
UDP server/socket is set.
net_on is called. net.server
net_on is called. net.server
net_start is called.
UDP port is set: 8000.
> net_socket_received is called.
Receive proto : UDP self_ref = 3 receive_ref = 1
userdata	 	romtable: 0x4023d118	

string	 	romtable: 0x4023c410	

test
udp packet sent completed 

dofile('udp_test.lua')
net_close is called.
net_create is called.
net_on is called. net.server
net_delete is called : net.server
 self_ref = -2
net_on is called. net.server
net_start is called.
UDP port is set: 8000.
> net_socket_received is called.
nud NULL 

```
FYI : A simular procedure with createServer/close on at TCP server works as expected

djphoenix, 2017-01-19:
Is this issue actual?

djphoenix, 2017-01-19:
Tested on latest `dev`, consider this is outdated -> close.

closed 2017-01-19 by djphoenix