nodemcu/nodemcu-firmware #234: net.socket:connect() - leaks memory on DNS lookup fail

Toshik, 2015-02-20:
```NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4```

net.socket:connect() leaks memory when called by hostname and DNS query failed. Base leak is about 240 bytes per call.

Test case:
```
function TestDNSLeak()
     c=net.createConnection(net.TCP, 0)
     c:connect(80, "bad-name.tld")
     tmr.alarm(1, 3000, 0, function() print("hack socket close, MEM: "..node.heap()) c:close() end) -- socket timeout hack
     print("MEM: "..node.heap())
end
```

Now run TestDNSLeak() every 5 seconds (do it not too fast, to get timer executed):

```
TestDNSLeak()
MEM: 20288
> DNS retry 1!
DNS retry 2!
DNS retry 3!
DNS retry 4!
DNS Fail!
hack socket close, MEM: 20840
TestDNSLeak()
MEM: 19952
> DNS retry 1!
DNS retry 2!
DNS retry 3!
DNS retry 4!
DNS Fail!
hack socket close, MEM: 20608
TestDNSLeak()
MEM: 19672
> DNS retry 1!
DNS retry 2!
DNS retry 3!
DNS retry 4!
DNS Fail!
hack socket close, MEM: 20336
TestDNSLeak()
MEM: 19456
> DNS retry 1!
DNS retry 2!
DNS retry 3!
DNS retry 4!
DNS Fail!
hack socket close, MEM: 20160
TestDNSLeak()
MEM: 19208
> DNS retry 1!
DNS retry 2!
DNS retry 3!
DNS retry 4!
DNS Fail!
hack socket close, MEM: 19936
TestDNSLeak()
MEM: 19008
> DNS retry 1!
DNS retry 2!
DNS retry 3!
DNS retry 4!
DNS Fail!
hack socket close, MEM: 19704
```

dvv, 2015-02-20:
It's not necessarily a leak as Lua performs garbage collection lazily.
Try calling `collectgarbage("collect")` before printing heap amount first. TIA

Toshik, 2015-02-20:
@dvv I have ran into that issue - my DNS was dead and ESP drains all memory...

Anyway, even with garbage collector same issue:

```
function TestDNSLeak()
     c=net.createConnection(net.TCP, 0)
     c:connect(80, "bad-name.tld")
     tmr.alarm(1, 3000, 0, function() collectgarbage("collect") print("hack socket close, MEM: "..node.heap()) c:close() end) -- socket timeout hack
     print("MEM: "..node.heap())
end
```

```
> TestDNSLeak()
MEM: 20072
> DNS retry 1!
DNS retry 2!
DNS retry 3!
DNS retry 4!
DNS Fail!
hack socket close, MEM: 20696
TestDNSLeak()
MEM: 19736
> DNS retry 1!
DNS retry 2!
DNS retry 3!
DNS retry 4!
DNS Fail!
> hack socket close, MEM: 20464
TestDNSLeak()
MEM: 19488
> DNS retry 1!
DNS retry 2!
DNS retry 3!
DNS retry 4!
DNS Fail!
hack socket close, MEM: 20184
TestDNSLeak()
MEM: 19264
> DNS retry 1!
DNS retry 2!
DNS retry 3!
DNS retry 4!
DNS Fail!
hack socket close, MEM: 20000
TestDNSLeak()
MEM: 19072
> DNS retry 1!
DNS retry 2!
DNS retry 3!
DNS retry 4!
DNS Fail!
hack socket close, MEM: 19808
```

mikewen, 2015-02-20:
This is related:
https://github.com/nodemcu/nodemcu-firmware/issues/189

If able to specify DNS servers, it would be helpful.
Google's DNS servers are much better than my local ISP's.

Suxsem, 2015-02-21:
I can reproduce this

MarsTechHAN, 2015-03-07:
The problem is at when SDK close a Socket...It will del it after a while(About 2hour)
[And they said each socket will cost 200bytes]

nodemcu, 2015-03-11:
try newest release [latest](https://github.com/nodemcu/nodemcu-firmware/releases/latest)

closed 2015-03-18 by nodemcu

UserXYZ, 2015-10-09:
This problem is still relevant, and I opened a new issue about it since I can't reopen this one...basically, the test used is the same like this here, and behaviour goes from full restart immediatelly, after a few dns retry/fail attempts or after some undefined time, like 10 minutes or more...

commit d28a2c9e (2015-03-11): add interger version release, fix #234, #252, #246