nodemcu/nodemcu-firmware #235: Sometimes no execution of LUA command (Version 20150213).

FourQuarks, 2015-02-20:
Every now and then the LUA interpreter don't execute the command only sending the prompt (see example at the bottom). It seems this occurs after a longer time (about 1/2 hour) without traffic from/to the WIFI interface.

> =status
2
> =status
2
> =status
2
> =status
>

FourQuarks, 2015-02-28:
This problem occurs also in version 20150214.

In the example below the line with the file.open is not executed. Therefore the next line with the file.write produces an error. 

> file.open("cs.txt", "w")
> file.write(string.char(0x00, 0x00, 0x00, 0x00, 0x00)) -- 20
stdin:1: open a file first
> file.write(string.char(0x00, 0x00, 0x5f, 0x00, 0x00)) -- 21 !
stdin:1: open a file first

marcelstoer, 2015-12-12:
See #719.

closed 2015-12-12 by marcelstoer