nodemcu/nodemcu-firmware #237: MQTT broken

biohazardxxx, 2015-02-22:
with regards to http://www.esp8266.com/viewtopic.php?f=24&t=1653&p=10417#p10417
The MQTT conncet does not seem to work anymore in latest firmware.

Mehrius, 2015-02-24:
Mehrius wrote:
I had the same problem. The problem is that, if loaded after boot through init.lua, the process is too fast and the chip is not connected to wifi when the script tries to publish or subscribe.

I fixed it by delaying the dofile in init.lua until the wifi.sta.status is 5. This is what my init.lua looks like and it works quite well.

This is my init.lua

if wifi.sta.status() ~= 5 then tmr.alarm(6, 10000,0, function(d) dofile('init.lua') end) return end
if wifi.sta.status() == 5 then dofile('mqtt.lua') return end

Hope this works for you...

tuanpmt, 2015-02-24:
Have you tried example application section in this thread: http://www.esp8266.com/viewtopic.php?f=19&t=1278

biohazardxxx, 2015-02-24:
I just did so and it does not work with NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4.
Let me know if you need any more info.

Mehrius, 2015-02-24:
Did you try my init.lua suggestion? 
It boots up and connects reliably for me on the same firmware...

mgf909, 2015-02-25:
I have the same issue. 
Even using Mahrius's init.lua it fails.

Mehrius, 2015-02-25:
This is my mqtt.lua which is called from init.lua.

<code>

require('ds18b20')
gpio.mode(4, gpio.OUTPUT)
ds18b20.setup(3)
ds18b20.read()
print(wifi.sta.status())
print('Setup done')

m=mqtt.Client(node.chipid(),120)
--m:lwt("/lwt","offline",0,0)
m:connect("192.168.0.106",1883,0, function(cnn) m:subscribe("/relay",0,function(m) print("Subscription done") end) end)
m:on("message", function(m,t,pl) if pl=='ON' then gpio.write(4,gpio.LOW) print(pl) end if pl=="OFF" then gpio.write(4,gpio.HIGH) print(pl) end end)

--m:on("offline", function(con) print("Reconnecting...") tmr.alarm(1,10000,0, function() m:connect("192.168.0.106", 1883, 0, function(conn) print("Connected") m:subscribe("/relay",0,function(m) print("Subscription done") end) end) end) end)

tmr.alarm(0,15000,1,function(test) m:publish("/temp",ds18b20.read(),0,0) end)

</code>

biohazardxxx, 2015-02-25:
Done as suggested without luck. I have added this files:
init.lua:
```lua
if wifi.sta.status() ~= 5 then tmr.alarm(6, 10000,0, function(d) dofile('init.lua') end) return end
if wifi.sta.status() == 5 then dofile('mqttTest2.lua') return end
```
mqttTest2.lua:
```lua 
print("Start")
m = mqtt.Client(wifi.sta.getmac(), 120)--, "user", "password")
m:lwt("/lwt", wifi.sta.getmac(), 0, 0)

m:on("offline", function(con) 
     print ("reconnecting...") 
     print(node.heap())
     tmr.alarm(1, 10000, 0, function()
          m:connect("192.168.0.1", 1883, 0)
     end)
end)

-- on publish message receive event
m:on("message", function(conn, topic, data) 
  print(topic .. ":" ) 
  if data ~= nil then
    print(data)
  end
end)

tmr.alarm(0, 1000, 1, function()
 if wifi.sta.status() == 5 then
     tmr.stop(0)
     m:connect("192.168.0.1", 1883, 0, function(conn) 
          print("connected")
          m:subscribe("sensors/test/temp",0, function(conn) 
               m:publish("sensors/test/temp","hello",0,0, function(conn) print("sent") end)
          end)
     end)
 end
end)
print("END")
```

The output after node.restart() is:
```
NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4
> Start
c_ÇÏRSöfJSúfJSúfêá

NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4
> Start
c_ÇÏRSöfJSúfJSúfêá

NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4
> Start
c_ÇÏRSöfJSúfJSúfêá

NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4
> Start
```

I have an ESP-12 with an DHT22 attached, powered by an own power supply.

biohazardxxx, 2015-02-25:
The same on 20150127 firmware works and gives:
```
NodeMCU 0.9.5 build 20150127  powered by Lua 5.1.4
> Start
END
connected
sent
sensors/test/temp:
hello
```

However on this firmware I have memory issues with the DHT22:
c_ÏRSöâFjSþâjSöfî<—þ‡$‡.,Š–Š‚ƒ,‡.,
–ˆ‡‡‡'@gá

NodeMCU 0.9.5 build 20150127  powered by Lua 5.1.4
> PANIC: unprotected error in call to Lua API (error loading module 'dht22' from file 'dht22.lua':
	not enough memory)

myplacedk, 2015-02-25:
```
NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4
> Start
c_ÇÏRSöfJSúfJSúfêá

NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4
>
```

I had the same problem yesterday, with very similar code. I'm guessing this block doesn't get executed, and makes it reset:
```
tmr.alarm(0, 1000, 1, function()
 if wifi.sta.status() == 5 then
     tmr.stop(0)
     m:connect("192.168.0.1", 1883, 0, function(conn) 
          print("connected")
          m:subscribe("sensors/test/temp",0, function(conn) 
               m:publish("sensors/test/temp","hello",0,0, function(conn) print("sent") end)
          end)
     end)
 end
end)
```

Try commenting out the line with ;publish, and the rest of the block will probably work.

I solved it by extracting the code to a function, something like this:

```
function onConnect(conn)
 if wifi.sta.status() == 5 then
     tmr.stop(0)
     m:connect("192.168.0.1", 1883, 0, function(conn) 
          print("connected")
          m:subscribe("sensors/test/temp",0, function(conn) 
               m:publish("sensors/test/temp","hello",0,0, function(conn) print("sent") end)
          end)
     end)
 end
end
tmr.alarm(0, 1000, 1, onConnect)
```
(Not tested.)

biohazardxxx, 2015-02-25:
nope. This:
```
print("Start")
m = mqtt.Client(wifi.sta.getmac(), 120)--, "user", "password")
function goConnect(conn)
 if wifi.sta.status() == 5 then
     print("status ok")
     -- tmr.stop(0) -- produces a crash
     print("timer stopped")
     m:connect("192.168.0.1", 1883, 0, function(conn) 
          print("connected")
          -- m:subscribe("sensors/test/temp",0, function(conn)
          print("subscribed") 
               -- m:publish("sensors/test/temp","hello",0,0, function(conn) print("sent") end)
          -- end)
     end)
 end
end
goConnect(conn)
```
it gives:
```
NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4
lua: cannot open init.lua
> dofile('script1.lua')
Start
status ok
timer stopped
> 
```

myplacedk, 2015-02-25:
This is a new situation, it no longer resets. Try adding an offline event receiver again, and see if that fires. Check the logs on the MQTT server.

biohazardxxx, 2015-02-25:
added
```lua
m:on("offline", function(con) 
     print ("reconnecting...") 
     print(node.heap())
     tmr.alarm(1, 10000, 0, function()
          m:connect("192.168.0.1", 1883, 0)
          
     end)
end)
```

result is
```
> dofile('script1.lua')
Start
status ok
timer stopped
> reconnecting...
7376
reconnecting...
7600
reconnecting...
7616
reconnecting...
```

myplacedk, 2015-02-25:
That means it tries to connect, but fails. For example invalid username/password or the server doesn't support MQTT 3.1.1. You need to check the MQTT server logs for clues.

biohazardxxx, 2015-02-25:
Installed a later version on my computer and there it works better (m:on("connect" does not seem to work? but doesn't matter now).
Connection and publish works. So it was the lack of MQTT 3.1.1 supported at my MQTT server that is necessary after the 20150127 firmware.
Sorry for that but thank you all for the guiding and help!
However I belive this thread will help others that stumble at this.

This is what the previous MQTT installation gave in the log:
1424900559: mosquitto version 0.15 (build date 2012-12-29 18:54:58+0000) starting
1424900559: Opening ipv4 listen socket on port 1883.
1424900559: Opening ipv6 listen socket on port 1883.
1424900565: New connection from 192.168.0.225.
1424900565: Invalid protocol "MQTT" in CONNECT from 192.168.0.225.
1424900565: Socket read error on client (null), disconnecting.
1424900575: New connection from 192.168.0.225.
1424900575: Invalid protocol "MQTT" in CONNECT from 192.168.0.225.
1424900575: Socket read error on client (null), disconnecting.

closed 2015-02-25 by biohazardxxx