nodemcu/nodemcu-firmware #238: Any way to distinguish a 'stable' version?

kireevco, 2015-02-22:
I would like us to have a clear link to a stable version. Proof of stability could be as simple as being around for long enough. We should have a "Download stable version" link in the main README.md.

marcoskirsch, 2015-02-22:
I think we should go slightly further and have several builds available, built with different options. For example, a build without floating point would be desirable since many applications don't require it but it takes a good chunk of memory.

dvv, 2015-02-22:
Are there _any_ unit tests? Without, any build is not reliable.

kireevco, 2015-02-22:
I personally think it would be great to have TravisCI set up. But I guess it's not possible without cross-compiling simplified. Any ideas?

myplacedk, 2015-02-22:
As far as I can see, it's a bit of a mess right now, but you can pick by selecting the right branch at get the latest pre_build from there.

master: pre_build/latest/nodemcu_latest.bin = pre_build/0.9.5/nodemcu_20150212.bin
dev096: pre_build/latest/nodemcu_latest.bin = pre_build/0.9.6/nodemcu_20150216.bin
dev: pre_build/latest/nodemcu_latest.bin = pre_build/0.9.5/nodemcu_20150213.bin

I guess it's supposed to be master=stable, dev###=beta, dev=bleeding edge. And if that doesn't work for you, build it yourself.

I think it's a great idea to have easy access to multiple downloads, such as stable/unstable, floating/integer.

dvv, 2015-02-24:
Well, dev is older than master atm... People, please, consider introducing at least some code housekeeping

GeorgeHahn, 2015-02-27:
I agree, it would be really cool to get this building on TravisCI. Travis supports running scripts - any reason not to pull down [pfalcon/esp-open-sdk](https://github.com/pfalcon/esp-open-sdk) for each build?

kireevco, 2015-02-28:
I would set up jenkins but I need more guidance for cross-compiling.
TravisCi doesn't support other C platforms.

GeorgeHahn, 2015-03-01:
Travis supports any toolchain you can load onto it. I *almost* had esp-open-sdk building on it, but gave up on a [weird error](https://travis-ci.org/GeorgeHahn/nodemcu-firmware/builds/52595368#L570) that went a bit over my head.

That said, I was able to build this repo using a prebuilt SDK: [Travis-CI](https://travis-ci.org/GeorgeHahn/nodemcu-firmware).

What do you guys think? If using a prebuilt SDK seems reasonable, I'll squash what I have right now and make a PR. Otherwise, I'll see if I can figure out how to get past the error I'm getting when building the SDK.

Once this is all working, someone with commit access to this repo can [setup Travis CI](http://docs.travis-ci.com/user/deployment/releases/) to push build artifacts from `master` as Github releases (and maybe builds from `dev` as well).

--

[Here](https://github.com/GeorgeHahn/nodemcu-firmware/blob/travis/.travis.yml) is the `.travis.yml` that works from a prebuilt SDK and [here](https://github.com/GeorgeHahn/nodemcu-firmware/blob/4dff36cf1c6900113657ae8f6bf5fc08b2f9bf87/.travis.yml) is the yaml that builds the firmware from scratch. The latter needs a couple of fixes (install pyserial & properly set PATH), both of which can be seen in the former yaml.

kireevco, 2015-03-01:
Wow, this is beautiful.

Definitely better than nothing.

It would be also nice to store binary files in a some good way, like
bintray generic package.

kireevco, 2015-03-02:
Can anyone test [nodemcu-firmware_v_8.bin](https://bintray.com/artifact/download/nodemcu/nodemcu-firmware/nodemcu-firmware_v_8.bin) it is built using [.travis.yml](https://github.com/kireevco/nodemcu-firmware/blob/travis/.travis.yml)

kireevco, 2015-03-03:
...Doesn't seem to work. I was wondering, what is a right way to merge 0x00000.bin and 0x10000.bin during the build process?

myplacedk, 2015-03-03:
I just pad 0x00000.bin with null-bytes until its size is 0x10000 (65536) bytes long, then add 0x10000.bin.

I don't know if it's correct, but it seems to work.

kireevco, 2015-03-03:
Thank you, how exactly o you pad? Do you have any piece of code for this?

devsaurus, 2015-03-03:
I use [srecord](http://srecord.sf.net) for that purpose:
```
srec_cat -output nodemcu_latest.bin -binary 0x00000.bin -binary -fill 0xff 0x00000 0x10000 0x10000.bin -binary -offset 0x10000
```

kireevco, 2015-03-03:
Thanks everyone for combining the knowledge. I was able to set up release process (in my [fork](https://github.com/kireevco/nodemcu-firmware) for now). I'd like someone from nodemcu/nodemcu-firmware repo managers to take a look and let us know what they think.

Thank you.

nodemcu, 2015-03-03:
pad with 0xff is great and right i think.
because i know little about CI, after google it, if this means build latest firmware after push. that would be great. help do this please.
and because many module rush into the repo, is there any chance that build customly by select which modules included?
PR welcomed, great job guys!

nodemcu, 2015-03-03:
seems that i should creat an acount in travis CI?
i will do this after i get to my pc.

dvv, 2015-03-03:
Missing is a set of at least smoke tests

nodemcu, 2015-03-03:
any ideas to run an auto test without a chip?

dvv, 2015-03-03:
not atm. but we could start at least with testing generic lua code on a host

kireevco, 2015-03-03:
I'll submit a pr for .travis.yml and you will need to add nodemcu-firmware repo to it. It will pick it up and start building. After you guys like a build. Add a git tag to latest commit of that new version and posh it to the repo - Travis will create a release with this tag. Let me know if that makes sense

kireevco, 2015-03-03:
Also, you will need to change a secure [key](http://docs.travis-ci.com/user/encryption-keys/):
Firstly remove current secure content (- secure in .travis.yml)
Secondly, 

```shell
gem install travis
travis encrypt GITHUB_OAUTH=<your_deployment_app_key_on_github> -r nodemcu/nodemcu-firmware -add
```
Or chat me in skype (kireev.co)

devsaurus, 2015-03-04:
Truely amazing!

Just a random though while reading through the Travis CI docs: Is it possible to deploy multiple binaries of the project in parallel? Like having a set of different firmware configurations that are built for a tag and releasing them alltogether.
At the moment users can download the full-featured pre_build binary and flash - all or nothing. Any customizations require them to modify `user_config.h` and recompile on their own. Different flavours of the firmware could address dedicated use cases.

Or, when heading for a more distributed approach: Can a novice user clone the repo, modify the firmware configuration (via .travis.yml or user_config.h?) and have it built on Travis CI? This would relief him from installing VMs & tool chains on his local computer. Saves much headache.

kireevco, 2015-03-04:
I was thinking about it. It would be nice to build float/non float version,
for instance. I need to test it. probably multiple - deploy sections will
work.

Someone, please set up travis-ci with nodemcu/nodemcu-firmware or give me
permissions, I'll do that.

nodemcu, 2015-03-05:
https://travis-ci.org/nodemcu/nodemcu-firmware

nodemcu, 2015-03-05:
is key secure thing a must?

kireevco, 2015-03-05:
Yes, otherwise it's not going to push to github releases (with tagging
magic and such)

"travis secure" protects your actual github application key

Make sure to create a separate github app key for travis

Use my file
https://github.com/kireevco/nodemcu-firmware/blob/travis/.travis.yml - no
reason to go through PR for one file.

GeorgeHahn, 2015-03-05:
The encrypted key is for Github OAUTH, which will allow TravisCI to deploy [releases](https://github.com/nodemcu/nodemcu-firmware/releases). It isn't essential, but it would be really nice to have.

GeorgeHahn, 2015-03-05:
@kireevco that .travis.yml needs a tar of the prebuilt tools too (tools/esp-open-sdk.tar.gz)

nodemcu, 2015-03-05:
ok I will get to a pc with ruby gem installed soon

kireevco, 2015-03-05:
Oh, that's right... It's in your .travis branch that I cloned from...

GeorgeHahn, 2015-03-05:
Yep, link is [here](https://github.com/GeorgeHahn/nodemcu-firmware/blob/travis/tools/esp-open-sdk.tar.gz).

We should probably store it in a separate repository though. Everybody who forks the repo shouldn't need to pull down a 35 meg file - and we probably shouldn't create the illusion that it's a good toolchain to use. We'll either get windows folks trying to use it or we'll have people trying to use it on Linux rather than building esp-open-sdk like they ought to. Essentially, it shouldn't be used for anything but the CI.

End goal is to have esp-open-sdk building on Travis, but putting the prebuilt toolchain into its own repository seems like a reasonable stopgap measure.

kireevco, 2015-03-05:
So feels like adding this as a first item in the install section should
work:
- wget
https://github.com/GeorgeHahn/nodemcu-firmware/blob/travis/tools/esp-open-sdk.tar.gz
-O tools/esp-open-sdk.tar.gz

GeorgeHahn, 2015-03-05:
Aha, perfect! :+1:

nodemcu, 2015-03-05:
hi, I finally get travis command line work.
and use
```
travis setup releases
```
and add on: tags
is it right?

nodemcu, 2015-03-05:
ok, seems work.
[releases](https://github.com/nodemcu/nodemcu-firmware/releases)
[nodemcu-firmware_v20150213.6.bin](https://github.com/nodemcu/nodemcu-firmware/releases/download/20150213/nodemcu-firmware_v20150213.6.bin)

kireevco, 2015-03-05:
No need to setup releases, just have in your .travis.yml:
```
deploy:
  provider: releases
  api_key: "${GITHUB_OAUTH}"
  file: "$TRAVIS_BUILD_DIR/bin/${file_name}"
  skip_cleanup: true
  on:
    tags: true
```

but make sure you set up a secure key:
```

travis encrypt GITHUB_OAUTH=<your_deployment_app_key_on_github> -r
nodemcu/nodemcu-firmware -add

```

nodemcu, 2015-03-05:
seems setup releases gen a key automatically, I have updated it to the .travis.yml. and it works.
thank you very much~

kireevco, 2015-03-15:
Can we please build last stable version and release it with 0.9.4/5 tag or would you rather wait until 0.9.6 will become stable?

nodemcu, 2015-03-16:
do you mean with name nodemcu_0.9.5_float_20150316.bin?
or without build date?

kireevco, 2015-03-16:
I would like to see:
All stable versions look like 1.9.5 or 1.9.5-stable, while dev should look like 1.9.9-dev-<date_or_build_counter>

This way it will be easy to download a stable version or get a dev one a shot.

All you need to do is to put a tag 1.9.5-stable to the commit that you would like to be built a bin from.

Hope it makes sense.

nodemcu, 2015-03-18:
all stable one: nodemcu_float_0.9.5_20150318
all dev one: nodemcu_float_dev20150317

closed 2015-03-18 by nodemcu