nodemcu/nodemcu-firmware #241: node.compile generate larger .lc than .lua

mikewen, 2015-02-23:
Is is normal .lc file size larger than .lua file ?
postip.lc size: 1044
postip.lua size: 993

Here is my code:
local localip=wifi.sta.getip()
local mac = wifi.sta.getmac()

--host="ipnodeusb.appspot.com"
host="i.nodeusb.com"

if (localip) then
     --local strPost = "GET /a?ip="..localip.."&mac="..mac.." HTTP/1.1\r\nHost: "..host.."\r\nConnection: keep-alive\r\nAccept: */*\r\n\r\n"
     local strPost = "GET /a?ip="..localip.."&mac="..mac.." HTTP/1.1\r\nHost: "..host.."\r\nAccept: */*\r\n\r\n"
     --local strPost = string.format([[GET /a?ip=%s&mac=%s HTTP/1.1\r\nHost: %s\r\nConnection: keep-alive\r\nAccept: */*\r\n\r\n]],localip,mac,host)
     print(strPost)
     
     local sk=net.createConnection(net.TCP, 0)

     sk:on("connection", function(sck) sk:send(strPost) end)

     sk:on("receive", function(sck, res) print(res) res=nil collectgarbage() end)

     sk:dns(host,function(conn,ip) sk:connect(80,ip) print(ip) collectgarbage() end)

     --print(localip)

     tmr.alarm(5,2000,0,function() sk:close() sk=nil strPost = nil collectgarbage() print(node.heap()) end)
else
     return
end

raz123, 2015-02-23:
AFAIK, there are no guarantees that the precompiled file will be of smaller size than its source file. From personal experiences, I found that only to be true after a certain threshold was reached.

Eg:
```
webide.lua        5285 bytes
webide.lc         2972 bytes
```

nodemcu, 2015-03-03:
compile only saves some ram
maybe waste more flash by increasing code size.

closed 2015-03-03 by nodemcu

GeoNomad, 2015-03-03:
Why are there so many 0x00 bytes in the .lc files? Sometimes as many as 24 in a row. No wonder the files get bigger.

FYI, Lualoader now supports uploading and downloading .lc files, which is very useful as we build things with multiple .lc files...

http://benlo.com/esp8266/index.html#LuaLoader

mikewen, 2015-03-04:
Even though size is large sometimes, use less heap though.

All in all, great feature, otherwise, I wouldn't be able to create my browser based IDE:
https://github.com/NodeUSB/nodemcu-ide