nodemcu/nodemcu-firmware #242: nodemcu on esp8266 does not "see" my wifi

andromedar, 2015-02-24:
Hi there..

I hope i'm in the right place here...

I am running a nodemcu-latest as of yesterday on an "01" module and the module does not  see my WiFi - but sees all the other networks that have far worse signal than mine. I tried changing channels on my router, changed b/g/n/auto settings...and it just won't work. All my other computers/phones work just fine. It is a TPLink 741 with OpenWrt firmware.

Anyone had similar issues?
Thanks

pipi61, 2015-02-24:
check Wireless Security Disabled/changed

andromedar, 2015-02-24:
I tried that already. I tried PSK, PSK2 and open...no luck :(

TheoHarbers, 2015-02-24:
I also had a problem with OpenWrt. My ADSL line was down for a couple of days but the LAN was up and running. The ESP got it's ip address from DHCP server on the OpenWrt but the ESP could not talk to other nodes on the lan.

andromedar, 2015-02-24:
Theo, thanks for your answer, but my problem is a little different in that i can not join my wlan at all, because my esp does not see it - but sees all other networks. Actually i tried now with a AT based firmware - and it doesnt work either. So, this is probably not a problem of nodemcu firmware...

inversesandwich, 2015-02-24:
Are you running a 5GHz wireless network? The ESP8266 only has a 2.4GHz radio.

andromedar, 2015-02-25:
No, it's not that either...i'm on 2.4G

inversesandwich, 2015-02-25:
Are you broadcasting your SSID?

andromedar, 2015-02-25:
Yes i am. 
Actually, the SSID is "OpenWrt".

inversesandwich, 2015-02-25:
The only other things I can think of is that you are using a wifi channel that isn't supported in some regions (I had this problem with a Raspberry Pi, Access point was using a channel that can be used in the UK but not in other countries and the WiFi dongle was set to a different region) or that it is a hardware issue seeing as it doesn't work with the AT firmware as well. I'd recommend getting another ESP module (preferable a different model altogether like an ESP12 in case there is a whole batch that's dodgy) before doing much else, then we can rule out any hardware issues.

andromedar, 2015-02-25:
Good idea :) I've seen that setting (Region) in OpenWrt. It is currently set to Slovenia, but can't remember if i also tried others. I didn't think that this could have any impact on channel frequencies. Will try that, thanks :)

andromedar, 2015-03-06:
UPDATE:  i flashed my TL-WR740N V4 from OpenWrt back to stock/factory firmware...and all seems to be ok. The ESP8266 is seeing my wlan just fine...

buzztiaan, 2015-03-12:
care to close the issue then? ;)

closed 2015-03-12 by andromedar

zikolach, 2015-04-29:
Same problem with soft AP on Raspberry PI using Realtek WiFi dongle. ESP cannot see SSID (some others are visible at the same time)

rm3nchaca, 2016-05-22:
I have the same problem with a tplink, I can connect manually, but I don't see it calling wifi.sta.getap().

At the first time I believed that was a "-" in the SSID because creating a portable access point with my phone called "test-8004" it wasn't shown. Then I remove the "-" and appeared, and also the tplink one.

Then when I turned off the "test", also the tplink disappeared!

I think that is something at parsing SSIDs.

Do you have any ideas of what can be happening?
Thanks a lot!