nodemcu/nodemcu-firmware #243: MQTT doesn't fire "connect" event

myplacedk, 2015-02-24:
According to the description in the API and the example in the API, there is two ways to do something when connected: Callback and "connect" event. But the connect even is never fired, I have to use the callback.

Is this an error in the code, or an error in the documentation?

dbzoo, 2015-03-01:
I noticed this too - there is some regression going on as release nodemcu_20150123 works fine.  In the meantime downgrade.

nodemcu, 2015-03-15:
could you give some code to test

dbzoo, 2015-03-15:
Assuming you have a MQTT server on that IP address.  For my test I was using mosquito.
In it simplest form
c = mqtt.Client("test.client", 120, nil, nil)
c:connect("192.168.1.8",1883, function(con)
     print("Connected")
end)

The word 'connected' is never printed because it doesn't.

mkarliner, 2015-03-23:
Bump. Any progress on this?

kireevco, 2015-03-25:
+1 same here. Works when done manually (no callbacks)

kireevco, 2015-04-05:
Thank you @nodemcu!