nodemcu/nodemcu-firmware #245: Bling example causes panic

emsi, 2015-02-24:
When I tried the blink examle the firmware got PANIC:

```
NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4

> lighton=0  tmr.alarm(1,1000,1,function()    if lighton==0 then       lighton=1       led(512,512,512)     else       lighton=0       led(0,0,0)     end   end)
> PANIC: unprotected error in call to Lua API (stdin:1: attempt to call global 'led' (a nil value))
```

GeorgeHahn, 2015-02-27:
It looks like you didn't define the `led` function as shown [here](https://github.com/nodemcu/nodemcu-firmware#do-something-shining).

nodemcu, 2015-03-03:
yes,
you have to define your led function.

closed 2015-03-03 by nodemcu

SkyRocknRoll, 2016-07-02:
Working Example
```
-- Blink using timer alarm --
timerId = 0
dly = 1000
-- use D4
ledPin = 4
-- set mode to output
gpio.mode(ledPin,gpio.OUTPUT)
ledState = 0
-- timer loop
tmr.alarm( timerId, dly, 1, function()
  ledState = 1 - ledState;
  -- write state to D4
  gpio.write(ledPin, ledState)
end) 
```
[Source](http://www.esp8266.com/viewtopic.php?p=19636)