nodemcu/nodemcu-firmware #248: http download API

mikewen, 2015-02-26:
It would be nice to add a http download function in net.
Something like:
net.download(url, fname)

Use lua to download a large file use lots of heap, and very easy to get 'out of memory' error.

dvv, 2015-02-26:
+1
I would vote pro the following contract (as downloading is essentially an async op)
```lua
net.download(url, function oncomplete(temp_fname, err)
  -- NB: temp_name chosen by node itself
  if err then
    tears_go_by(err)
  else
    -- atomically replace target file
    file.rename(temp_fname, target_fname)
    -- go on
  end
end)
```

nodemcu, 2015-03-03:
not appropriate to put http download function api in net module.
maybe some one make some http module. and do this like http.get(url, ..)

Manawyrm, 2015-03-03:
I've already written that and my code does that: https://github.com/Manawyrm/ESP8266-HTTP

dvv, 2015-03-03:
https://github.com/Manawyrm/ESP8266-HTTP/blob/master/http.lua#L18 -- this simply will run node out of memory, should a file be of any valuable size.

Manawyrm, 2015-03-03:
that is the wanted behaviour of http.lua. 
It allows you to get small returns like api responses directly as a string.

If you want to download big files, theres httpDL.lua which directly loads into the spi-flash ;)

dvv, 2015-03-03:
Right.

Please, consider:
- using printf-style `("fmtString"):format(args)` instead of tons of `..` to save memory.
- `str:op(...)` instead of `string.op(str, ...)`
- not regexp-based flavor of `find`: `inputbuffer:find(pattern, 1, true)`

Manawyrm, 2015-03-03:
Thanks for the advices. 
I will try to implement the changes after I get home :)

TerryE, 2015-11-02:
If some offers a module then we will consider this, but the current contributors will not be offering this.

closed 2015-11-02 by TerryE