nodemcu/nodemcu-firmware #249: Can a software error cause a freeze (not reboot)?

Suxsem, 2015-02-26:
Hi all!
In my init.lua I have
```
srv = net.createServer(net.TCP)
srv:listen(80, function(conn) 
	conn:on("receive",function(conn,payload)
		conn:send("ok")
	end)
	conn:on("sent",function(conn) conn:close() end)
end)
.....................................
```
and then A LOT of other code.
Sometimes (maybe after 10 minutes, maybe after 10hours) the esp8266 stops to work and doesn't reply to port 80 anymore. I have to power cycle it.

My question is: can a software error of my other code cause FREEZE or software errors only cause REBOOT?
I'm on 13/02/2015 versione of nodemcu

nodemcu, 2015-03-03:
when it's not reply to port 80,
try 
```lua
wifi.sta.disconnect()
wifi.sta.connect()
```
to see if the connect to router is lost?

TerryE, 2015-11-03:
See #719

closed 2015-11-03 by TerryE