nodemcu/nodemcu-firmware #252: tmr.time() says that my device is on from 2414 days :D

Suxsem, 2015-02-28:
Hi guys, today i tried the tmr.time() function on nodemcu 2015-02-13 and it says 208624296 (and counting)...
it's more then 2414 days :D

maybe....a bug?

frankexoo, 2015-02-28:
Hi,
The return value is in us, not seconds.
And it's a uint31, so will overflow every 35 minutes.

Suxsem, 2015-02-28:
Hi. thanks for your reply.
But...are you sure is us?
Because seems to me is seconds. Can you test?

frankexoo, 2015-02-28:
Hi,
Yes I'm sure.
Please check the API description.
You can test by reading it twice with some seconds in between: you will see the update in the highest digits.

Suxsem, 2015-02-28:
from API:
```
tmr.time()

Description

return rtc time since start up in second, uint31 form.
```
i'm not at home now, i will test later

frankexoo, 2015-02-28:
Oops, I was referring to tmr.now()...
But I quickly tested tmr.time() with the same result: it returns us, not seconds.
Not sure what the difference is between the two though.

eliabieri, 2015-02-28:
whats happens when it overflows?

-- 

Elia Bieri
whateveryouwant@eliabieri.com

frankexoo, 2015-02-28:
After an overflow the timer will start at 0 again.

Update: checking once more now shows that tmr.time() does return time in seconds since power-on.
My previous test may have been caused by a strange behavior of the ESP8266: when using tmr.time() instead of tmr.now() in a script caused my ESP8266 to crash during download and running a of script.
So some improvements regarding stability seems necessary.

So that also leaves the initial question unanswered....

Suxsem, 2015-02-28:
Thanks for your test. I think I will you tmr.now() for now...

frankexoo, 2015-02-28:
Hi, it seems I'm experiencing the same defect as you.
In a web-application I was using tmr.now(). I switched to tmr.time().
The first couple of hours the numbers returned were OK. But after a few hours it is now at 208622239.
This is remarkably in the same range as you found.
So I think we can safely say this is a bug.

Suxsem, 2015-02-28:
Ok great. Did you encour any issue using tmr.now() / 1000 ?

frankexoo, 2015-02-28:
No. Seems to be working correctly.

Suxsem, 2015-02-28:
I just remembered that it will overflow every 35 minutes...so not a replacement for tmr.time().
i hope it will be fixed

MarsTechHAN, 2015-03-07:
Beacuse the SDK document just mentioned we have this function instead of any explanation(They are always like this at first... )We have no idea about how to use it...So...For now try to find the truth by yourselves : )

nodemcu, 2015-03-07:
really? have you review the code of this part and debug for this bug?

nodemcu, 2015-03-11:
try newest release [latest](https://github.com/nodemcu/nodemcu-firmware/releases/latest)

closed 2015-03-18 by nodemcu