nodemcu/nodemcu-firmware #255: Example server code freezes the ESP8266

marcoschwartz, 2015-02-28:
Hi, I am using the example code to run a server on the ES8266:

srv=net.createServer(net.TCP) 
    srv:listen(80,function(conn) 
      conn:on("receive",function(conn,payload) 
        print(payload) 
        conn:send("<h1> Hello, NodeMcu.</h1>")
      end) 
      conn:on("sent",function(conn) conn:close() end)
    end)

It works fine once, but freezes on the second request. I am using the latest version of the 0.9.5 firmware. Can you help on this issue? Thanks!

mikewen, 2015-03-01:
Could be the problem of browser keep connection open. (Think your server is http 1.1)
And the second time, it just send the data without sending headers, and your code already close the connection.

marcoschwartz, 2015-03-01:
Thanks! I've been thinking about something like that, but how can I solve it? I mean, this is 100% the example code from the README file, so it should work right?

MarsTechHAN, 2015-03-07:
If we put http-head inside... The whole things will be very complex, because most the browser won;t act like that, so we think it is a trade-off that we like to see.

nodemcu, 2015-03-11:
Hi, I am using Chrome. the example works fine. could you test more?

marcoschwartz, 2015-03-12:
It works fine for me as well, but as I said it freezes the ESP8266 after the first request. Does that part works fine for you? Thanks

nodemcu, 2015-03-12:
yes, can do many requests, I press re-fresh button in Chrome.

dvv, 2015-03-12:
you should try responding with minimum correct HTTP response (see 1.0 for client to not keep the connection open):
```lua
conn:send("HTTP/1.0 200 OK\r\n\r\n<h1>Hello, NodeMcu.</h1>")
```

mercury123456, 2015-03-16:
I have the exact same problem.
I found that setting STATIONAP insted of STATION at the beginning the connection is always alive, I know is not the right way but it works fine. I will find a better solution in the future.

mercury123456, 2015-03-16:
found here:
http://www.esp8266.com/viewtopic.php?f=6&t=1946

TerryE, 2015-11-03:
See #719

closed 2015-11-03 by TerryE