nodemcu/nodemcu-firmware #261: Separate repo for lua examples & contrib?

kireevco, 2015-03-03:
Sounds like it would be beneficial to let people contribute their lua code freely to a separate repo, like nodemcu/nodemcu-contrib or nodemcu/nodemcu-lua

eliabieri, 2015-03-03:
+1
I think this is a very good idea

-- 

Elia Bieri
whateveryouwant@eliabieri.com

marcoskirsch, 2015-03-03:
Instead of having a separate repository with many examples, I suggest keeping a markdown document with links to repositories of interesting examples.

Sourcing the examples here causes too much off topic noise. A single repository with lots of examples would have a similar problem.

dvv, 2015-03-03:
Central contrib repo would require either a relevant and concerned moderator or would drift toward pulling everything they request to pull.
I would instead spawn a wiki page where people could maintain a list of modules, just like they do in node.js, say.

kireevco, 2015-03-03:
I don't disagree.
Question. How rich is the list of good lua/esp8266 repos to link to right
now?

marcoskirsch, 2015-03-03:
Here's one I wrote: https://github.com/marcoskirsch/nodemcu-httpserver
General purpose web server with support for server-side scripting.

dvv, 2015-03-03:
@kireevco I believe it is well enough for starters to just call the people to put their links here. Then you as initiator will compile the first version of the list ;)

kireevco, 2015-03-03:
I like links in the wiki idea.
Although, current wiki has to be refactored, and homepage should not be a
changelog.
Ideally it should be nothing more 1-2 pages glossary. Thoughts?

dvv, 2015-03-03:
I vote pro wiki. But this issue might be accumulating the first contributions, as people imo look rather at issues not changes to wiki. So to collect the first page of modules this I personally consider the right place.

dvv, 2015-03-03:
This is my humble contrib: https://github.com/dvv/nodemcu-thingies

marcelstoer, 2015-12-12:
With #772 the examples in this repository will be updated. #771 or #774 might have room for links to other good examples.

closed 2015-12-12 by marcelstoer