nodemcu/nodemcu-firmware #262: compile with lua mathlib support

makefu, 2015-03-03:
Hi,
i tried to compile nodemcu with mathlib support but it seems like commenting in the lines which hint towards mathlib in `app/lua/linit.c` and `app/lua/lualib.h` is not enough.

when running `make` i get the following error:

    lua/.output/eagle/debug/lib/liblua.a(lmathlib.o):(.text.math_tan+0x0): undefined reference to `tan'
    lua/.output/eagle/debug/lib/liblua.a(lmathlib.o): In function `math_tan':
    /opt/Espressif/nodemcu-firmware/app/lua/lmathlib.c:59: undefined reference to `tan'
    ...

maybe i am a bit too naive and i am missing something here :)

nodemcu, 2015-03-06:
1, you should also add "-lm" in your makefile:
```
	-lssl	\
	$(DEP_LIBS_eagle.app.v6)					\
	-Wl,--end-group			\
	-lm
```

2, make sure libm.a is can be found. ( I copy this file from windows vm toolchain to lubuntu toolchain)

3, disable(comment out) most function in lmathlib.c,  otherwise it will too big to fit in region `iram1_0_seg', which can not be expand.

nodemcu, 2015-03-11:
math lib now supported in dev branch.

closed 2015-03-11 by nodemcu