nodemcu/nodemcu-firmware #268: Add Ultrasonic sensor driver

mikewen, 2015-03-07:
How about add Ultrasonic sensor driver ?

EAD Fritz already wrote C code:
https://github.com/eadf/esp8266_ping

MarsTechHAN, 2015-03-07:
Too many block will make the whole program get bigger and bigger which is what we don't want to see. 
Can you rewrite that in Lua ? Or wait we add the ‘block select’ feature.
(Also... We highly recommend you to just pull for our people are always in busy...)

mikewen, 2015-03-07:
I wrote the function in Lua , but because Ultra Sound sensor need very accurate timing, it was not working well.

nodemcu, 2015-03-07:
you can make a PR to dev branch, 
and currently disable the module. for those who need this module, pull and rebuild it.

dvv, 2015-03-07:
Wonder if we can load binary modules like big Lua does with  .so files?

inversesandwich, 2015-03-09:
@dvv You can compile a Lua script into a binary, however it often doesn't make it much smaller, it just reduces the RAM usage.

dvv, 2015-03-09:
I meant namelt binary modules not compiled scripts. That resembles
intrinsic modules like bit e.g.

TerryE, 2015-12-12:
This falls into the won't fix unless we can get a contributor to develop this and open a PR.

closed 2015-12-12 by TerryE