nodemcu/nodemcu-firmware #270: UDP Broadcast reception is unstable

tizbac, 2015-03-07:
The nodemcu firmware will not always receive udp broadcasts, especially after a while it has been on, when it stops working, packets with it's ip address or ICMP still work

pastukhov, 2015-04-06:
Can you show an example?

tizbac, 2015-04-06:
https://github.com/tizbac/IoTManager/blob/master/esp8266/init.lua

If you send commands to it using a broadcast socket , from another host, after some time , it will randomly stop working

marcelstoer, 2016-08-07:
Feel free to reopen if you still see those symptoms with a recent 1.5.x firmware.

closed 2016-08-07 by marcelstoer

max-weller, 2017-09-16:
Hi, I'm seeing exactly the same behaviour with NodeMCU version 2.1.0

The minimal sample code I'm using is as follows:

```
wifi.sta.config({ssid="...",pwd="..."})
wifi.sta.connect()

tmr.create():alarm(3000, 0, function()
print('IP: ',wifi.sta.getip())

scsock = net.createUDPSocket()
scsock:listen(2020)

scsock:on("receive", function(scsock, l, port, ip)
    print("received udp from ",ip,port)
end)
end)
```

I use python on my PC to send the UDP broadcasts / unicasts:

```
import socket,time
UDP_PORT=2020
UDP_BCAST_ADDR="255.255.255.255"
MESSAGE=b"Hello, world."
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM) # UDP
sock.setsockopt(socket.SOL_SOCKET, socket.SO_BROADCAST, 1)  
sock.bind(("0.0.0.0", UDP_PORT))
while True:
    sock.sendto(MESSAGE, (UDP_BCAST_ADDR, UDP_PORT))
    time.sleep(1)
```

For ~10 seconds after resetting the nodemcu, all packets come through. Afterwards, only occasionally a packet is received.

max-weller, 2017-09-16:
Interestingly, if I send the broadcasts at fast pace (10 packets/second), it works for a longer time (~50 seconds).