nodemcu/nodemcu-firmware #278: If mqtt is trying to connect to a down broker will cause nodemcu restart.

crapthings, 2015-03-14:
How to prevent this ?

mqttClient:on("offline", function(conn)
	print("Reconnect to broker...")
	--mqttClient:close()
        if(wifi.sta.status()==5) then
	    mqttClient:connect(brokerSettings.addr, brokerSettings.port, 0)
        end
end)

mkarliner, 2015-03-14:
Would putting the connection in a pcall help?

I use this to stop restart on publish

tmr.alarm(1,10000,1, function()
	temp = (adc.read(0)-500) /( 195/10)
     print(temp)
     if pcall(readtemp) then
          print("Temp sent OK")
     else
          print("Temp err" )
     end
	end)

function readtemp() 
      m:publish("temperature",temp,0,0, function(conn) print("sent") end)
 end

nodemcu, 2015-03-14:
which one cause restart?

crapthings, 2015-03-14:
@nodemcu 

mqttClient:connect(brokerSettings.addr, brokerSettings.port, 0)

@mkarliner 

will try it thanks

nodemcu, 2015-03-14:
from the c code, currently, if a mqttClient is offline( tcp disconnected)
it will release resources, and become a nil ( seems not right though)
so, for re-connection, try to create a client again.