nodemcu/nodemcu-firmware #290: Node will never get IP address if lua script started emedialy after node.restart()

evgenitznikolov, 2015-03-17:
Hi, 

While playing with the module I noticed quite inconsistent behavior - the node was not getting an IP quite often. After some experiments I am quite sure that the reason was that I was executing my test script too fast after the reset (using node.restart()). 

Here is my test script:

```LUA
print (wifi.sta.status())
print (wifi.sta.getip())

if (wifi.sta.getip()==nil) then 
  wifi.setmode(wifi.STATION)
  wifi.sta.config("my SSID","my Password")
  while (wifi.sta.status()~=5) do  tmr.wdclr() print(wifi.sta.status()) end
end
```

Obviously it is not a showstopper but I don't know what will happen if I put this in init.lua...

Regards, 

E.

nodemcu, 2015-03-17:
1, getip will return nil if it is called immediately after ssid setup. because ap takes time to assign a ip to station.
2, 
```lua
while (wifi.sta.status()~=5) do  tmr.wdclr() print(wifi.sta.status()) end
```
this code will prevent other part(tcp/ip stack) from running, wifi.sta.status() will never change.  makes  this a forever loop.
3, use tmr.alarm() to check this periodically.

evgenitznikolov, 2015-03-17:
Does tmr.delay() also prevent the tpc/ip stack to run? I assumed it is using an interrupt but apparently it doesn't?

nodemcu, 2015-03-17:
yes.
tmr.delay() is same as while loop. for loop
it uses cpu all the time.

closed 2015-03-17 by evgenitznikolov