nodemcu/nodemcu-firmware #291: on("connection", ...) event doesn't seem to get fired

evgenitznikolov, 2015-03-17:
Hi, 

I made this test script and I never get "connection" event executed. Do I do something wrong?!

```LUA
    print(wifi.sta.getip())
    srv=net.createServer(net.TCP) 
    srv:listen(6066,function(conn) 
      conn:on("receive",function(conn,payload) 
        print("Payload: "..payload) 
        conn:send("-\n")
      end) 
      conn:on("sent",function(conn) print("<S>") end)
      conn:on("connection",function(conn) print("<C>") end)
      conn:on("reconnection",function(conn) print("<RC>") end)
      conn:on("disconnection",function(conn) print("<DC>") end)
```
The "disconnection", "sent" and "receive" events work like charm.

What is the meaning of "reconnection" - it doesn't seem to fire either but I don't know how to achieve this. 

Regards, 

E

nodemcu, 2015-03-17:
```lua
srv:listen(6066,function(conn) <----this is the same as on:("connection".

...
conn:on("connection",function(conn) print("<C>") end)
-- when calling this, actually conn is already connected. 
...
```

evgenitznikolov, 2015-03-17:
Thank you! I am fairly new to this environment, hence the silly questions!

closed 2015-03-17 by evgenitznikolov

snikul, 2017-01-31:
I register one more undocumented callback on an unrecognized event when I play with nodemcu-httpserver by marcoskirsch. Neither of 5 documented events(request, sent ,connect ,disconnect , reconnect  ). Payload of it = nil. What event evokes that callback? I used old debug print method.
                   .  .  .
local function onConnect(connection, payload)
            print( "Connection" )
         end
local function onReConnect(reconnection, payload)
            print( "ReConnection" )
         end
         print( "conn reconnection" )
         connection:on("reconnection", onReConnect )
         print( "conn receive" )
         connection:on("receive", onReceive)
         print( "conn sent" )
         connection:on("sent", onSent)
         print( "conn disconnection" )
         connection:on("disconnection", onDisconnect)
         print( "conn connection" )
         connection:on("connection", onConnect )
         print("Callback on Unrecognized Event")
         print( payload )
     end
   )
   return s
********************************** Esplorer Report
For example, when I send 192.168.137.144 from Chrome I get the following Esplorer Report.
                       .  .  .
------------------------------------------Unrecognized event below
conn reconnection
conn receive
conn sent
conn disconnection
conn connection
Callback on Unrecognized Event
nil                                                                                                     <----- **payload**
-------------------------------------------192.168.137.144/index.html "request" event
GET: /index.html
Finished sending: 	http/index.html
-------------------------------------------Unrecognized event below
conn reconnection
conn receive
conn sent
conn disconnection
conn connection
Callback on Unrecognized Event
nil
-------------------------------------------
GET: /underconstruction.gif
--------------------------------------------
I would appreciate Your friendly comments! Thanks a lot. Serge