nodemcu/nodemcu-firmware #292: restart without error message

kubi57, 2015-03-17:
Lua retarts without any error message with the following example code:

s="%3F"
s = string.gsub(s, "%%(..)", function(a) return string.char(tonumber(a, 16)) end)
print(s)

Am I doing anything wrong, or is this a bug?

nodemcu, 2015-03-18:
you are passing a function to string.gsub as the third parameter?

kubi57, 2015-03-18:
Yes, that wors fine. If you put 65 insted of tonumber(a, 16) it replaces the found patterns with the letter A. So functions as third parameter do work, but another function in there is apparently too much.

kmpm, 2015-03-22:
I have something similar...
```
> s='abc'
> -- this works
> print(s:gsub('.', function (c) return c:byte() end))
979899  3
> --this crashes shortly after the result is printed.
> print(s:gsub('.', function (c) return string.format('%02X', c:byte()) end))
616263  3
>�S�٦N�;;W���

NodeMCU 0.9.6 build 20150318  powered by Lua 5.1.4
...
```

kmpm, 2015-03-22:
This fails as well..
```
> s='abc'
> print(s:gsub('.', function (c) return string.char(97) end))
aaa     3
> J�
    @�M�Ds9ɗ����
```

kmpm, 2015-03-22:
The unhex function in https://github.com/nodemcu/nodemcu-firmware/issues/59#issuecomment-77480952 fails as well.

model101, 2015-04-09:
Moin,

at least the given examples should be fixed in dev branch now.

Fix was reducing stack usage of string functions.

Carsten

devsaurus, 2015-12-13:
No issues seen with the examples above on current firmware.
See #719.

closed 2015-12-13 by devsaurus