nodemcu/nodemcu-firmware #296: Error build nodemcu-firmware

CHERTS, 2015-03-18:
I use to build <a href="http://programs74.ru/udkew-en.html">Unofficial Development Kit for Espressif ESP8266</a>, 2 days ago all was compiled successfully, but now this error.

To build the files I use the following console.cmd:
```
@echo off
title Build nodemcu-firmware
SET ESP8266SDK=C:\Espressif\
PATH=%ESP8266SDK%xtensa-lx106-elf\bin;C:\MinGW\bin;C:\MinGW\msys\1.0\bin;%PATH%
%comspec%
```
Next steps, I run console.cmd and execute commands make, 2 days ago firmware successfully compiled, now comes the error:

```
xtensa-lx106-elf-gcc  -L../lib -Wl,--gc-sections -Xlinker -Map=mapfile -nostdlib
 -T../ld/eagle.app.v6.ld -Wl,--no-check-sections -u call_user_start -Wl,-static
-Wl,--start-group -lc -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lmain -ljson -lsm
artconfig -lssl user/.output/eagle/debug/lib/libuser.a driver/.output/eagle/debu
g/lib/libdriver.a lwip/.output/eagle/debug/lib/liblwip.a json/.output/eagle/debu
g/lib/libjson.a upgrade/.output/eagle/debug/lib/libupgrade.a platform/.output/ea
gle/debug/lib/libplatform.a libc/.output/eagle/debug/lib/liblibc.a lua/.output/e
agle/debug/lib/liblua.a coap/.output/eagle/debug/lib/coap.a mqtt/.output/eagle/d
ebug/lib/mqtt.a u8glib/.output/eagle/debug/lib/u8glib.a smart/.output/eagle/debu
g/lib/smart.a wofs/.output/eagle/debug/lib/wofs.a spiffs/.output/eagle/debug/lib
/spiffs.a cjson/.output/eagle/debug/lib/libcjson.a modules/.output/eagle/debug/l
ib/libmodules.a -Wl,--end-group -lm -o .output/eagle/debug/image/eagle.app.v6.ou
t
c:/espressif/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../
xtensa-lx106-elf/bin/ld.exe: .output/eagle/debug/image/eagle.app.v6.out section
`.text' will not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
make[1]: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1
make[1]: Leaving directory `/d/Neo/Project/ESP8266/Project/nodemcu-firmware/app'

make: *** [.subdirs] Error 2
```

nodemcu, 2015-03-18:
lib/function not from the source code, like printf/strcat and most standard lib func.  goes into iram1_0_seg section, which can not be expanded.
maybe the toolchain have a diff standard lib in your situation.
try to remove some api/module.  for example, some function in math module.

CHERTS, 2015-03-18:
What build environment you use? What operating system do you use to build the firmware?

nodemcu, 2015-03-18:
now the firmware is build in travis-CI.
and the environment is esp-open-sdk
ref this: https://github.com/nodemcu/nodemcu-firmware/blob/master/.travis.yml

devsaurus, 2015-03-18:
Do you get app/mapfile from the failing gcc call? We could compare this against a mapfile from a successful linker run.

CHERTS, 2015-03-18:
I disabled the math module and firmware successfully compiled.

File user_modules.h 
```
//#define LUA_USE_BUILTIN_MATH		// for math.xxx(), partially work
```