nodemcu/nodemcu-firmware #299: Problem writing bin file to ESP8266 and different error message

JohnSmith-LT, 2015-03-18:
Hi,

Most likely something I am doing but using esp8266_flasher to write the integer build binary nodemcu_integer_0.9.5_20150318.bin results in this message:

Writing at 0x00065c00... (100 %)
Failed to write to target Flash

Which is not the same as when you write the ESP_8266_BIN0.92.bin and get:

Writing at 0x0007ec00... (99 %)
Leaving...
Failed to leave Flash mode

which seems to work fine...

Any thoughts? Is there another flasher that I should use?

nodemcu, 2015-03-18:
https://github.com/nodemcu/nodemcu-flasher

JohnSmith-LT, 2015-03-18:
Thank you. You know i've actually used that before as well but completely forgot about it! D'oh

JohnSmith-LT, 2015-03-18:
Actually, upon just trying https://github.com/nodemcu/nodemcu-flasher it seems that the 64bit build doesn't work on my 64bit Win7... Anyone else seeing this with the latest version updated 3 days ago?

nodemcu, 2015-03-18:
I pull it down, works ok
I'm using 64bit win7 too.

JohnSmith-LT, 2015-03-18:
Hmm thats strange... Thanks for trying it though and confirming... I'm using Win7 virtualised on a mac but that shouldn't make any difference... i'll investigate.

closed 2015-03-18 by JohnSmith-LT