nodemcu/nodemcu-firmware #300: Can't get client ip with conn:getpeername()

cracker0dks, 2015-03-18:
I get an error if I call conn:getpeername() on TCP Server side. This sould work i guess?! I Cant get the client IP which is connected.

nodemcu, 2015-03-19:
what is the error?
ip,port = conn:getpeer()

cracker0dks, 2015-03-19:
Error on "conn:getpeername()": PANIC: unprotected error in call to Lua API (stdin:2: attempt to call method 'getpeername' (a nil value))

Error on "ip,port = conn:getpeer()": PANIC: unprotected error in call to Lua API (init.lua:9: attempt to call method 'getpeer' (a nil value)

The code is: -- A simple http server
    srv=net.createServer(net.TCP)
    srv:listen(80,function(conn)
      conn:on("receive",function(conn,payload)
        ip,port = conn:getpeer()
        print(ip)
        conn:send("YEAH!")
      end)
      conn:on("sent",function(conn) conn:close() end)
    end)

cracker0dks, 2015-03-19:
Sry, maybe this was an old bug (flashed 10 Days ago). I flashed the new firmware (updated 5 days ago) and now this thing is running: ip,port = conn:getpeer() THANKS :)

closed 2015-03-19 by cracker0dks