nodemcu/nodemcu-firmware #302: ipv6 support (by updating lwip)

flokli, 2015-03-18:
According to http://www.esp8266.com/viewtopic.php?f=21&t=1347, IPv6 support should be relatively straightforward by updating the lwip library.

momorientes, 2015-04-22:
+1 IPv6 would be a great feature

TerryE, 2015-06-06:
The thing that concerns me here is that until Espressif come out with Rev B silicon with 2x or 4x the RAM, if we keep adding features, then we might end up with a wonderfully featured firmware that can only run a 10 line application.

Personally the last thing that I wan't is my IoT devices directly addressable on the Internet.  Mine will always run in a privarte address space and any external or end-user access will  be through a properly secured HA layer.

flokli, 2015-06-06:
@TerryE Sure, but IPv6 ULAs are also a much nicer way for private IP ranges. Additionally, you can route them from trusted devices ;-)

FreshXOpenSource, 2015-06-06:
... and - in a proper setup - control via a firewall what gets thru to the device. i guess in a modern "cloud" / IoT world ipv6 is a mandatory feature.

marcelstoer, 2015-06-06:
> i guess in a modern "cloud" / IoT world ipv6 is a mandatory feature

It only becomes a mandatory feature when IPv4 is widely not supported anymore. Or if the intended context in which a system is used requires it. Currently, IMO neither of those applies to ESP8266/NodeMCU.

flokli, 2015-06-06:
According to [RFC6540](https://tools.ietf.org/html/rfc6540), IPv6 already is mandatory for all IP-Capable Nodes.

Additionally, new IoT protocols like [6LoWPAN](http://en.wikipedia.org/wiki/6LoWPAN) suggest enabling IPv6 support even more. So I strongly suggest to consider adopting it!

marcelstoer, 2015-06-06:
RfC6540 is not a standard but merely a best practice documentation. Nothing wrong with that, I don't think anybody disagrees with RfC6540...

> This memo documents an Internet Best Current Practice.

TerryE, 2015-06-07:
@flokli, The IPv6 stack will probably leave the current ESP8266 with little or no available RAM.  On the next generation of IoT devices with 256KB RAM it will be a good idea, but so would decent security and a more functional OS.

cal101, 2015-06-07:
@TerryE Do you have any foundation for that claim? 
We are not talking about an IPV6 router, just a client.
A quick search in the internet made be hope that for basic stuff it's <200 bytes more for lwip IPV6.

TerryE, 2015-06-07:
@cal101 Carsten, we just need someone to prepare the patch, preferably with a user define to give users the option of disabling it.  We can do the sizing then.  At this growth rate on the firmware size, we'll have no room let for Lua apps in a year's time :unamused:

cal101, 2015-06-07:
Sorry, not me at the moment.

MarsTechHAN, 2015-08-09:
Espressif change some function of LWIP and it is no more opensource now. To change it will be hard job, when comparing to the good that ipv6 give us, that not worth.
And I believe Espressif will support IPv6 when it is necessary.

Aeprox, 2015-08-23:
Just for reference: espressif recently [released an open-source patch of lwip](http://bbs.espressif.com/viewtopic.php?f=46&t=951). They say it's "based on SDK 1.3.0", that might be a problem? Still could be worth investigating by someone more knowledgeable on the subject.

cralli, 2015-09-18:
So more than 20% of hits to Google from the USA and Germany are over native IPv6 and you guys say that your firmware/sw will not support it (against the suggestion of the IETF for IP capable stacks)..

The unique reason for IPv6 today in digital/Internet products is business continuity.

Good luck!

For those interested, I keep more Internet6 stats/info here: 
 http://the-internet6.blogspot.com.es

TerryE, 2015-09-18:
@cralli, IMO you are missing an important issue here and that is security.  This generation of IoT devices simply does not have the capacity or capability to run a secure OS and S/W stack.  Anyone who attaches this sort of device or their fridge or garage door to the public Internet deserves any hacking or malicious attack that they open themselves up to.

Again IMO, the only sensible way to open these devices up to the public internet is through some firewall / bastion tier which contains the HA system running on a properly configured and locked-down Linux instance.  In this use case the devices will be running on a private LAN in a private address space, so IPv6 is hardly an issue.

At the moment we are struggling the SDK and firmware into this device and still leave enough RAM/Flash available to implement a minimal real-world application.  If you want IPv6 then come back in 3-4 years when the devices have the capacity to run a proper IoT OS which has all of the necessary security in place.

TheSkorm, 2015-09-26:
NAT isn't a security measure :)  any consumer router should be dropping inbound connections from the internet on v6 anyway. 

IPv4 subnets are at a premium now. 

Agree on the memory constraints though. Be nice if it were an option to turn on and off and same with the v4 stack. 

The great thing about v6 and IoT is that you don't need to handle NAT punching to get two IoT things to talk to each other over the internet. 

IoT not the private subnet of things.

TerryE, 2015-09-26:
I entirely agree that NAT is not a security measure in its own right.  My point is that any sensible security architecture using current technology would firewall and proxy through an application tier such as a properly configured home automation system so you may as well use a private address space for your IoT devices.  

The idea of allowing any IoT to present itself on the Internet give me the willies.  Feel free to do this work and issue the pull request.  I am not interested in helping.

eyaleb, 2015-09-26:
@TerryE , I am 100% with you here.

I am constantly shocked to see how people who should know better actually are comfortable with cloud services, trusting everything to it. And now same people want to expose their devices to the same (unknown) cloud. If we learnt anything in the last few years it is that there is <b>no security</b> and <b>no privacy</b> on the 'net. We call is the "cloud" but we should really call it "public arena". Everyone sees the convenience and is blind to the cost.

It started by people making their whole life public, now they want to make it physically public.

The fact on the internet is: when you allow access to one (e.g. yourself, accessing your home devices) you allow everyone.

And remember: just because I am paranoid does not mean they are not after me...

Back on topic: now that we are all agreed on the above :-) , there is no reason to have IPv6 in the house.

cheers

wnagele, 2015-09-26:
This discussion is odd. How people are implementing things is not something you can control by providing or not providing a network stack. People are going to build/have built stupid shit and only once that creates issues will there be progress towards more secure systems. If you want to change that you have to change the way our current market economy works.

The question here is simple - can a IPv6 stack be made to work on this architecture (maybe not even dual stack with IPv4). If so - it's worth doing, cause future and stuff. If it's something that is not possible due to restrictions in the hardware - well then that's that. Other products will appear that have sufficient capability and will serve that need.

TerryE, 2015-09-27:
See #625.  We need a current SDK to do this and getting one is proving tighter that a duck's a\*\*e.

marcelstoer, 2015-09-27:
At this point the discussion has become pretty pointless for me to continue and I suggest to close this issue. IMO:
- Some would benefit from IPv6 support, others couldn't care less.
- For the majority of use cases there's nothing you can do with IPv6 you can't do with IPv4.
- None of the core contributors is interested in this feature and, therefore, won't spend any time on it. So, unless someone provides a PR this won't be implemented regardless of how many upvotes it gets.
- IPv6 support is best addressed at the SDK level.
- Sooner or later i.e. by upgrading to a recent SDK we'll get this feature automatically.

nickandrew, 2015-10-03:
:-1:  Close. Fully discussed :-)

TerryE, 2015-10-03:
Suggest that we revisit this if and when we have a chipset and build that has sufficient resources to support it.

closed 2015-10-03 by TerryE

ffrediani, 2015-10-18:
What a shame, whatever reason this is not going ahead. Either because some people can't care less (and don't understand the importance of IPv6 to the IoT world) or because the device "might" not have suitable resources for that.
I understand by this discussion that some people don't even believe, don't want to understand or don't want to give a try to make it happen, which doesn't help with the progress and search for a proper solution. This is certainly not a good approach for improvement of any kind. Make up their mind before even trying.

TerryE, 2015-10-18:
@ffrediani, we have been working on the sdk 1.4 based release which include the open LWiP.  So instead of moralising, and telling us what to do ot what not to do, why don't *you* fork the repo and preprare, test and propose the patch yourself?

marcelstoer, 2015-10-18:
> don't understand the importance of IPv6 to the IoT world

Importance is subjective. Don't assume that because other people don't share your priorities they don't _understand_. Over and out.

ffrediani, 2015-10-18:
Folks, I never said it was "my priorities", but that IoT has everything to do with IPv6, and that giving up without trying isn't something good.
Imagine OpenWrt project if people had said in the past that was impossible to have a working and amazing Linux distribution running in just 4MB of flash memory !

TerryE, 2015-10-18:
@ffrediani, actions not words.  If you feel that strongly, then put in the effort and do it yourself.  IMO, IoT and IPv6 are orthongal.

jmattsson, 2015-10-27:
Ignoring hardware resource constrains, another issue is that LWIP is *really* tightly tied to the SDK, and ships as a binary library by default. When doing the upgrade to 1.4.0 I ended up having numerous emails back and forth with Espressif to sort out various issues in this are, and Espressif released updates and fixes along the way.

A large change such as introducing IPv6 would require a substantial amount of ongoing maintenance as well, and would make SDK upgrades more challenging. Great if someone wants to do it, even better if that someone is Espressif (assuming the hardware can cope, that is).

nickandrew, 2015-10-27:
IPv6 support would be great but I think resources are too constrained for it at present.

What I'd rather see first is an ultra-stable platform, one which does not experience random watchdog resets and "MEM CHECK FAIL" errors.

jmattsson, 2015-10-27:
@nickandrew The old "MEM CHECK FAIL" is but a memory now after the SDK upgrade! Espressif has replaced it with the much less dire sounding text "don't use rtc mem data". It was never an actual error as far as I understand, it just indicated that the SDK didn't find any stashed away calibration/what-not data in RTC memory on boot.

I completely agree it sounded scary though - the first time I failed to flash a NodeMCU firmware onto a fresh board and only saw that message I thought I'd broken my hardware!

nickandrew, 2015-10-27:
@jmattsson My device, built from https://github.com/nickandrew/nodemcu-firmware/tree/onewire-power started rebooting 10 times a second with that error - but not always, and believe it or not - placing a finger on the can often made it stop. Is that the typical behaviour of the error the SDK fixed?

grahamehorner, 2015-10-27:
just to add my two pennies +1 for ipv6 +3 for stable platform including HTTPS client support as 90% of the services I require to communicate with are HTTPS. I'm looking into HTTPS/SSL/TLS issues across a number of project using the epsressif ask in hope to bottom out what I think is an sdk bug.

marcelstoer, 2015-11-06:
> we did enable IPv6 in the ESP8266 RTOS SDK! and it will be in ESP32 as well.
https://twitter.com/EspressifSystem/status/662658821307797504

ardje, 2016-01-31:
I know this is closed. And will be opened again when sdk supports it.
I only wish for ipv6 link-local to work, since I then always know the address of the device.
Furthermore, it would relieve the system of tasks like dhcp, and other administrative tasks related to ipv4. So the ipv6 stack can be more lightweight than the v4, with optional adding public addresses.
This would bring the system on-par with 6lowpan devices which only link-local addresses.

danrl, 2016-05-05:
+1 for IPv6. +1 for TLS over IPv6.
 Make it optional if you think that makes it more appropriate for the common user. I'd love to be able to disable IPv4 stack, to save RAM. I don't use IPv4 in my IoT network anyway. Also, maybe we can make IPv4 optional as well?

Sorry for posting in a closed thread.

jonathanmbradshaw, 2016-08-04:
+1 for IPv6 

Perhaps I have miss understood IPv6/IPSec but TLS (unless TLS is a requirement at the application layer) is unnecessary in IPv6. Transport layer encryption for network connections is available in the IPv6/IPsec stack.  All the Best JMB

pjsg, 2016-08-04:
In practice, people don't implement IPSEC in IPv6 (you have all the key management problems of TLS to worry about).

marcelstoer, 2016-08-04:
It may incidentally be related to #1435 as Espressif appears to have included IPv6 in SDK 2.0.

jmattsson, 2016-08-05:
They did?? I didn't see a mention of that. Do we have any RAM left now?

marcelstoer, 2016-08-05:
> They did?? I didn't see a mention of that.

Espressif's "John Lee" sort of confirmed that on Twitter (2.0 announcement).

TerryE, 2016-08-05:
IIRC, the standard LWiP open-source has configuration defines to enable IPv6 compilation.  If Espressif wave picked up this version then we should at least be able to make IPv6 support configurable if users don't require it and don't want to take the RAM hit.

synnack, 2016-11-11:
Good to hear. At least make IPv4 configurable too, because why take the RAM hit on a protocol from the 1980s that's completely obsolete these days?

TerryE, 2016-11-11:
> a protocol from the 1980s that's completely obsolete these days?

I agree that it's obsolescent, but it's not going to be obsolete until the majority of home routers and ISPs support IPv6 as standard.  I know it's my personal opinion, but IMO anyone who allows external Internet access for their IoT devices (now or for the foreseeable future) has no idea of the security threat of doing so.  IPv4 on a private subnet will do me for the next 10 years, at least.

pjsg, 2016-11-11:
I don't know which of the various v4/v6 transition technologies are supported by LWIP. As time marches on, v6 will become dominant and may become the only native transport (my cellphone only has v6 connectivity native (tmobile) and uses a transition technology to support v4).

I'm using lwip-2.0.0 on another project which is 99% v6 and it works well.

ffrediani, 2016-11-14:
Obsolete it is not. It is old, and will stay around for at least a decade. Move towards IPv6 is always great, but remove support for IPv4 isn't certainly a good thing. So just leave both options as optional.

nickandrew, 2016-11-14:
Much as I prefer IPv6 whenever possible, no way is IPv4 obsolete.

washichi, 2017-03-18:
Is ipv6 working on the NodeMCU? if so, how can I use it?
(I want to use it to solve all the port forwarding hassle)

SpComb, 2017-04-20:
With the 2.0 release and "Net functionality rewrite on top of LWIP (#1379)", which no longer seems to use the IPv4-only `espconn` layer, would `net` module support for IPv6 now be more feasible than before?

digits122, 2021-03-31:
+1 
I think it's time to reopen this issue now.I want IPv6 but not IPv4, we should leave IPv4 or IPv6 as a module to choose, or even other net layer protocol by developer.
My network has IPv6 public address full support ,but I have only Symmetric NAT IPv4 address, that means its no way to build p2p network for IPv4, but with IPv6 everything is possible.

ardje, 2021-05-02:
Exactly. I don't want dhcp and stuff like that. IPv6 link local is the only thing I want. And it will clear a lot of issues. And if your browser is incapable of IPv6 link local, it's a bug in the browser, that's easy to circumvent with socat.

pjsg, 2021-05-02:
It looks as thought the IDF supports ipv6. It would seem that the easier approach would be to make the lua network layer understand v6 on the esp32 branch.