nodemcu/nodemcu-firmware #303: Stack size problem

4refr0nt, 2015-03-19:
I wrote a module 'nkBMP180.lua' in the 2.5 kb and only 108 lines.
When I load it my test: require 'nkBMP180.lua ",
I get the message:
> dofile('test_BMP180.lua')
error loading module 'nkBMP180' from file 'nkBMP180.lua':
" not enough memory"
---------------------------
This message indicates a stack overflow ( see http://lua-users.org)
This is not a problem heap, the stack size is a problem.
It is necessary to increase the size of the stack in the following assemblies nodeMCU.

model101, 2015-04-09:
Moin,

thank you for esplorer! It's a helpful tool!
Can you show us the lua file in question?
The error message does not indicate a stack error, but a heap error.
The stack size can't be increased easily if at all. AFAIS the standard stack
upper limit is used 0x3fffffff and currently no lower limit is set which will
result in reboots if memory is trashed - typically the SDK times list - and the watchdog timer kicks in.
So what we need for stability is less stack usage, not more.
If you show your example in the esp8266 forum more people could discuss the matter.

Carsten

TerryE, 2015-11-02:
The current `dev` issue addresses this.

closed 2015-11-02 by TerryE