nodemcu/nodemcu-firmware #305: Can't recover from PANIC unprotected error in call

kwave, 2015-03-22:
Lets assume I accidentally uploaded a faulty init.lua which (e.g.) calls a function on a nil pointer. I will get a "PANIC: unprotected error in call to Lua API (attempt to index a nil value)" and nodemcu keeps rebooting, which will produce the error once more.
Flashing a new init.lua is not possible due to the error messages interrupting the transmission process.

The only way to recover is to re-flash nodemcu. There should be a possiblity to upload a new script, even with faulty init.lua

nodemcu, 2015-03-22:
reflash.
and make a init.lua:
```lua
--init.lua, something like this
countdown = 3
tmr.alarm(0,1000,1,function()
	print(countdown)
	countdown = countdown-1
	if countdown<1 then
		tmr.stop(0)
		countdown = nil
		if file.open("user.lc") then
			file.close()
			dofile("user.lc")
		elseif file.open("user.lua") then
			file.close()
			dofile("user.lua")
		else
			print("user.lc and user.lua not exist")
		end
	end
end)

```

then other code goes into user.lua.

mkarliner, 2015-03-22:
Can you tell what is wrong with this alternative?
No-one seems to be using pcall to catch exceptions.
Is there something wrong in doing this?

  if(pcall(runmain)) then
          print("Running OK")
  else
          print(err )
  end

 function runmain() 
        dofile('user.lua')
 end

nodemcu, 2015-03-22:
well, your way is better way. the lua way.
```lua
--init.lua, updated.
countdown = 3
tmr.alarm(0,1000,1,function()
    print(countdown)
    countdown = countdown-1
    if countdown<1 then
        tmr.stop(0)
        countdown = nil
        local s,err
        if file.open("user.lc") then
            file.close()
            s,err = pcall(function() dofile("user.lc") end)
        else
            s,err = pcall(function() dofile("user.lua") end)
        end
        if not s then print(err) end
    end
end)
```

kwave, 2015-03-22:
I would expect that such a "error loop" is handled by nodemcu without explicitly implementing a recovery behaviour. Something like "if restart cause was N-times a panic, do not execute init.lua anymore until device is repowered"?

timoline, 2015-04-04:
+1 for kwave,
please provide a dummyproof bootstrap,

chadouming, 2015-04-04:
The "dummyproof bootstrap" you linked won't help at all if you flashed a
faulty init.lua

timoline, 2015-04-04:
If you use the init.lua above, it will never fails...

But you are correct, once a faulty init.lua, you have to reflash

jorgempy, 2015-05-17:
on a related matter, if i have a PANIC in a program my ESP01 doesn't reboot anymore, it prints garbage and that's it, power cycling doesn't help, 'esptool.py run' doesn't help, the only thing that helps is to reflash the 'esp_init_data_default.bin'

ideas?

BesitzeRuf, 2015-05-17:
If its printing garbage, mby try other baudrates?

jorgempy, 2015-05-17:
output of picocom (each time there's garbage i power cycled the ESP)
what other baudrates can I try?: 
```
picocom /dev/ttyUSB0 
picocom v1.7

port is        : /dev/ttyUSB0
flowcontrol    : none
baudrate is    : 9600
parity is      : none
databits are   : 8
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,

Terminal ready
��
  t�(�h�4�@>�t��$�x��|��z6C�
*** baud: 4800 ***
�Lr��R���4�d��
*** baud: 2400 ***
A�MEȵ�i�
*** baud: 1200 ***
����
*** baud: 600 ***
�~�
*** baud: 300 ***

*** baud: 600 ***

*** baud: 1200 ***

*** baud: 2400 ***

*** baud: 4800 ***

*** baud: 9600 ***
��E����h�͍H<
           $���4C$��`~4b�
*** baud: 19200 ***
��a�qy����%!!��q�a��Q�!Tc���aE�����Ǆ�y�!�����PaEa�
*** baud: 38400 ***
@�
  q�Ȩ����^Ѫ 9�J$½V(�B(��9�(�ąc������JȞ�3(%���ʉ�
��/������� ��D��v*&��C                         )ğv補ᄦDÇ
                      Ȩ�Ȏz�|
*** baud: 57600 ***
GM�X�@	eI��E���#Q@��]_[��Ϋ�Tօ�a\�eHY@��eIuK����dJ�����Y��ޅH��D��a�C�YH��aLqFa���@G�����Y��ޅH�D��
                                                                                                                  a�C�AJQ@��e��iFa���AG����H�Ć�/�!aF`���F?
*** baud: 115200 ***
{ll��|�d�|

            �
             d�
               b<���s�cd
                          #��no�$gn���cp�lcdsdp�'�
                                                       �l

                                                           
                                                            c
                                                              n�|��l���'o�d�`'o
                                                                                    $`gs�o
                                                                                            rc$`p�'�
                                                                                                         s�ܜ��
                                                                                                                co�<�
                                                                                                                        
                                                                                                                         c��g'�
                                                                                                                                l �gn
                                                                                                                                         d`gs�n
                                                                                                                                                 �
                                                                                                                                                   d`p�g�r�������

         c
          '�|$l����gn�l`�gnl�$`'{�ol r��g
                                                       
                                                        l`r�
*** baud: 230400 ***
��������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������
*** baud: 460800 ***
������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������
```

jorgempy, 2015-05-21:
Sounds related to #387 maybe a panic puts more load on the chip? to be continued...

TerryE, 2015-12-12:
Going back to the OP, if the Lua developer / the application does an unprotected call which errors, then this should error and reboot.  If the developer wants to prevent this then the standard Lua constructs apply.   This is a feature, not a bug.

closed 2015-12-12 by TerryE