nodemcu/nodemcu-firmware #306: compile() api does not overwrite existing lc

kmpm, 2015-03-22:
If you compile a lua file using the api it looks as if it's doing it's work but it doesn't overwrite the existing lc file.
Is this as it should be? If so, please make a note in the api docs.

eyaleb, 2015-03-23:
I do not see this. I often update a `*.lua` file then `node.compile()` it and I always see the changes reflected in `dofile("*.lc")`. At times going through this cycle for dozens of iterations.

Maybe your file system is broken? Clear it with `file.format()` and re-upload the programs then check if this happens.

devsaurus, 2015-08-11:
@kmpm Your report reads like you hit a real issue back then. But it's quite old and SPIFFS was updated several times in the meantime. Could you please check whether you still face this issue with current `dev` branch? Like @eyaleb, I never encountered such behaviour and can't reproduce.

devsaurus, 2015-09-05:
Feel free to reopen in case the issue persists.

closed 2015-09-05 by devsaurus