nodemcu/nodemcu-firmware #307: Nodemcu-flashed chip hangs after some (over?)load.

mgarmash, 2015-03-22:
It's strange but somehow the esp8266 under some load hangs and never back online. I can connect to it, but it does not response any command.

I read some similar symptoms here https://primalcortex.wordpress.com/2015/02/13/esp8266-nodemcu-and-mqtt-event-subscription-load-testing/

>One interesting thing was that after some testing not even doing a reset or completely powering it off, I was able to bring the esp8266 back online. It just put out garbage on the serial output and that was it. Strange indeed.
>So I’ve reflash it again with the latest NodeMcu version (lots of reflashing hey  ), but still the esp8266 did not recover! Only garbage out when connecting to the serial port (And no, it wasn’t the serial port rate issue). So I’ve reflashed again the original AT Firmware, and it came back to life again, and then reflash it again with the latest Nodemcu version.

This guy and me invented some similar way to fix the problem. Two continuous reflashes 
1) AT-firmware (normal response after that step)
2) Nodemcu-firmware

This code is guaranteed to put my chips to such "frozen" state after some time. 
https://github.com/mgarmash/esp8266-hacks

May be this links also can be helpful: 
http://www.esp8266.com/viewtopic.php?f=23&t=1404&start=10
(Russian language)
http://esp8266.ru/forum/threads/nodemcu.10/page-3#post-1444
One interesting message is that "it's probably solved going back to 2015-01-08 build"

mgarmash, 2015-03-22:
I'm continuously getting this "frozen" situation in different variants. For example running some file uploading during timer-event execution. 

BTW. I'm using latest (https://github.com/nodemcu/nodemcu-firmware/releases/tag/0.9.5_20150318) nodemcu build.

4refr0nt, 2015-03-22:
if frozen state do you have LED light half strong on ESP?

mgarmash, 2015-03-22:
After power up the red-led flickers a bit and the blue-one blinks one time. Cant's measure now if red-led is half strong after that. Will try "froze" some chips and compare with normal ones :)

4refr0nt, 2015-03-22:
I'm have some troubles. For resolve - all GPIO need pullup with
resistors (gpio15 pulldown)

marvinroger, 2015-03-22:
I can confirm, I have this problem too. The blue LED blinks one time on power-up, then doesn't blink anymore; the red one, however, is constantly on.

4refr0nt, 2015-03-22:
If blue LED blink on boot, then module feel good. Check connection to
USB-TTL. When you successfuly send command and receive answer from ESP
blue LED blink too.(Red LED - power indicator)

marvinroger, 2015-03-22:
Well, I had an init.lua on the ESP8266 that connected to the Wifi. The fact is, when the chip "hanged", the connection was not made anymore after power-up, and I was not able to communicate with it using my USB-TTL. The only solution I found, as said above, was to flash the AT firmware and flash NodeMCU back, and then it worked again - without changing connections to USB-TTL -.

4refr0nt, 2015-03-22:
NodeMCU firmware has smallest size (not 512k), and do not have blank.bin for resizing. Reflash with AT firmware (real 512k) - simple way to brick repair

nodemcu, 2015-03-23:
hi, can you try file.format() after re-flash nodemcu?
it seems that the file system part of the flash corrupted.
maybe I should add format to the c code?

mgarmash, 2015-03-23:
@4refr0nt that is the answer for continuos reflashing, thx!
@nodemcu doing file.format in "hanged" state is not possible. Chip is not responding to commands at all. Only reflashing can bring it to life again. BTW I was suggested to try some hardware-hacks (pull-up resistors, VCC capacitors and may be shorting wires)...

marvinroger, 2015-03-23:
This personally happens with an ESP8266-01, don't know if that might help.

marvinroger, 2015-03-23:
Okay, I managed to get the module out of "hanged" state by flashing the `esp_init_data_default.bin` at 0x7C000 instead of flashing with the whole AT firmware. It is a 128 bytes file containing:

`0500 0402 0505 0502 0500 0405 0504 0505
04fe fdff f0f0 f0e0 e0e0 e10a ffff f800
f8f8 524e 4a44 4038 0000 0101 0203 0405
0100 0000 0000 0200 0000 0000 0000 0000
e10a 0000 0000 0000 0000 0193 4300 0000
0000 0000 0000 0000 0000 0000 0000 0000
0000 0000 0000 0000 0000 0000 0000 0000
0000 0000 0000 0000 0000 0000 0000 0000`

And the fact is it resurrects the module...

rickyok, 2015-03-28:
Confirmed. Happened to me too. Have to reflash everytime it freeze.

dmcinnes, 2015-03-30:
Having the problem too. Sometimes I'm able to get it restarted if I turn it off and on a bunch of times. Changing the GPIO connections seems to help too. Also on rev 1 hardware with the latest firmware.

kr1os, 2015-04-03:
Same issue here.  As soon as I upload any init.lua and do a reboot, I can not communicate with the esp until I reflash the AT firmware.

marvinroger, 2015-04-06:
How do you all wire the chip-select pin (CH_PD)? Using a resistor or raw 3.3V?

model101, 2015-04-09:
Moin,

has anyone a short stripped down example that does not need the whole internet or
elektronik lab to be reproduced?

Carsten

mgarmash, 2015-04-10:
Just FYI, I'm wiring all connections without any resistors (should try it).

eyaleb, 2015-04-10:
FWIW my flash dance is
`  0x00000 /path-to/blank512k.bin 0x00000 0x00000.bin 0x100000x10000.bin`
So the first 512KB is always cleared.

YMMV

model101, 2015-04-10:
Moin,

I would like to find out what part gets corrupted and prevents booting.
I took a look at the esp8266 boot loader. The boot loader message is written very early
to the uart (serial). It's something like

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 31972, room 16 
tail 4
chksum 0x28
load 0x3ffe8000, len 2852, room 4 
tail 0
chksum 0x54
load 0x3ffe8b30, len 14572, room 8 
tail 4
chksum 0xd0
csum 0xd0

It is written on startup of my esp 01 using 74880 baud, so set your terminal accordingly.
If you semi-bricked your esp, what does it report on reset using the baud rates 74880/115200/8900 baud?

Carsten

vlast3k, 2015-08-12:
Mine is a an ESP01 module, with the nodemcu firmware "NodeMCU 0.9.5 build 20150318  powered by Lua 5.1.4"

The dump was

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 32028, room 16 
tail 12
chksum 0xd7
ho 0 tail 12 room 4
load 0x3ffe8000, len 2888, room 12 
tail 12
chksum 0xef
ho 0 tail 12 room 4
load 0x3ffe8b50, len 15348, room 12 
tail 8
chksum 0xc5
csum 0xc5

The module was working fine for more than 4-5 months. During this time it was turned on the whole time. Yet i got a problem with Lua hanging when the memory is low, and memory getting low when i try to connect to a web page and my internet is down.
So i decided to turn it on by keeping CH_PD high, sending the data, and then setting CH_PD low. This happens in 2 minute cycles. 2 times it survived the internet outage successfully. Yesterday it died, and i needed to flash esp_init_data_default.bin at 0x7C000. The setup was that there is internet connection to the Router, but the router is connected to a network that redirects to a login page. I am not sure if this is the problem, i will try to reproduce it in the same setup in the next few days. Hopefully it will break again

TerryE, 2015-12-12:
See #719

closed 2015-12-12 by TerryE