nodemcu/nodemcu-firmware #309: Filesystem bug for used space more than 40-50k

4refr0nt, 2015-03-23:
1. Format filesystem file.format()
2. do code:
for j=1,3 do
  print("20K"..j)
  file.open("20K"..j..".txt", "w")
  for i=1,1280 do
    file.writeline('0123456789ABCDE')
  end
  file.close()
  print("file 20k"..j..".txt created")
end
3. chek file list:
----------------------------
20K1.txt        : 20480 bytes
20K2.txt        : 20480 bytes
20K3.txt        : 13568 bytes
----------------------------
Total file(s)   : 3
Total size      : 54528 bytes

Total : 68021 bytes
Used  : 55722 bytes
Remain: 12299 bytes
4. Try with j=1,4 and j=1,5

vowstar, 2015-03-27:
Confirmed, will fix it soon.

pellepl, 2015-03-30:
What is the bug? That the filesystem becomes slower or that there are remaining bytes?

raz123, 2015-03-30:
@pellepl, my understanding is that the code tries to create 3x 20 kb files, and while 2 files are created of that size, the third one doesn't quite make it, despite the fact that there is plenty of space remaining. Some slowness is also seen as a side effect of the "bug".

pellepl, 2015-04-05:
Sorry for belated answer, this could actually be something bad. Not a bug probably, but suboptimal. I'll analyse it further as soon as I can.
As for the slowness: this is due to spiffs is working against a pretty full filesystem, and thus needs to do more garbage collecting = moving pages and erasing blocks = takes time.

pellepl, 2015-04-05:
Been poking around a bit, made GC a bit more admissive. The 20K3.txt will at least hit 19185 bytes with the patch before fs is full. Still, free space is reported as 9287 bytes, even though were fat. No more can be done (without creating spiffs2).

The problem is that when creating files in this manner, the file headers need to be updated once in a while. After writing the three files, in NodeMCU setup, this leads to that almost every block will have 2 deleted pages and the rest of pages written. The deleted pages are included in the report of the "free" space - although not actually free, the garbage collector is supposed to be able to free them and are thus included in the non-used bytes when reporting free space.

But - and here's the but - when having only two deleted pages per block, and the with files like these, spiffs needs around two extra pages when erasing a block and rewriting the file lookup pages of those files residing in that block. So, a garbage collection will free 2 pages, but as file data is moved, 2 new pages are needed for pointing out the new locations, and the garbage collector will come up freeing nothing.

If anyone understands above rambling, that is why there seems to be free space but there actually isn't. There simply is an unfortunate pattern on the filesystem.

Anyhow, patch soon to come.

raz123, 2015-04-05:
@pellepl, thanks for looking into this! :)

pellepl, 2015-04-07:
@raz123, no prob :) The patch is in.

raz123, 2015-04-11:
@vowstar, can you update nodemcu with the latest spiff changes?

raz123, 2015-04-13:
@pellepl, I notice that some file name corruption can happen if the hardware restarts at the wrong time. It results in files with names such as the following. Is there a way to delete these files, other than by using their file names? (doesn't work)

```0 !",t`      : 1447 bytes

pellepl, 2015-04-13:
Hmm.. I would run a SPIFFS_check.
It could be potentially dangerous trying to remove these files, as they
probably are not files but either file data or inode data being erroneously
marked as files. This can happen when a garbage collect is interrupted by a
power cycle.

SPIFFS_check will sanity check all metadata and clean things up.

Is it possible to get a dump of the state? Preferably before you run check
;)

Cheers / Peter

4refr0nt, 2015-05-11:
With 4Mbyte (32Mbit) module we have only 1664kb for files:
```text
> file.format()
format done.
> 
----------------------------
No files found.
----------------------------
> 
Total : 3441461 bytes
Used  : 0 bytes
Remain: 3441461 bytes

> for j=1,24 do
print("128K"..j)
file.open("128K"..j..".txt", "w")
 for i=1,8192 do
file.writeline('0123456789ABCDE')
end
file.close()
end
128K1
128K2
128K3
128K4
128K5
128K6
128K7
128K8
128K9
128K10
128K11
128K12
128K13
128K14
c_ПRSцfJSъfJSъвоЎCб
NodeMCU 0.9.6 build 20150406  powered by Lua 5.1.4
lua: cannot open init.lua
> 
Total : 3441461 bytes
Used  : 1738677 bytes
Remain: 1702784 bytes

> 
----------------------------
128K1.txt       : 131072 bytes
128K10.txt      : 131072 bytes
128K11.txt      : 131072 bytes
128K12.txt      : 131072 bytes
128K13.txt      : 131072 bytes
128K14.txt      : 15360 bytes
128K2.txt       : 131072 bytes
128K3.txt       : 131072 bytes
128K4.txt       : 131072 bytes
128K5.txt       : 131072 bytes
128K6.txt       : 131072 bytes
128K7.txt       : 131072 bytes
128K8.txt       : 131072 bytes
128K9.txt       : 131072 bytes
----------------------------
Total file(s)   : 14
Total size      : 1719296 bytes

```

pellepl, 2015-05-13:
For 4 meg sizes, and if you're running default nodemcu spiffsconfig (4096
byte logical block size, 128 byte page size) this is not very optimal
configuration. Try 65536 byte logical block size and 256 byte page size.

TerryE, 2015-11-03:
See #719

closed 2015-11-03 by TerryE