nodemcu/nodemcu-firmware #310: URLdecode function causes reset

mrvanes, 2015-03-23:
I'm trying to parse a multiline incoming POST body var and have managed to reduce my testcase to the following code:
```
v="abc%0D%0Adef"
print(string.gsub(v, "%%(%x%x)", function(x) return string.char(tonumber(x, 16)) end))
```
The ESP prints out correctly the decoded string
```
abc
def    2
```
but then resets itself.
nodemcu version: nodemcu_integer_0.9.5_20150318.bin

nodemcu, 2015-03-24:
bug confirmed and don't know why, hard to debug...
the weird thing is 
```lua
v="abc%0D%0Adef"
print(pcall(function() print(string.gsub(v, "%%(%x%x)", function(x) return string.char(tonumber(x, 16)) end)) end))
-- result: true, means no error, and no reset.
```
currently you can use pcall to test and run the chunk.
```lua
v="abc%0D%0Adef"
pcall(function() print(string.gsub(v, "%%(%x%x)", function(x) return string.char(tonumber(x, 16)) end)) end)
```

model101, 2015-04-09:
Moin,

smells like the stack overflow problem which becomes obvious with the LUA_BUFFERSIZE
of 4 K which is reduced in latest dev release.

Carsten

anusoft, 2015-06-08:
Ugly code that can percent-decode ascii

```
local unescape = function(s)
    local rt = ""
    local skip = 0
    for i = 1,#s,1 do
        local c = s:sub(i,i)
        if skip > 0 then
            skip = skip - 1
        else
            if c == "%" then
                hex = s:sub(i+1,i+2)
                skip = 2
                rt = rt .. string.char(tonumber(hex, 16))
            else
                rt = rt .. c
            end
        end    
    end
    return rt
end
```

TerryE, 2015-06-08:
I can see that you want to avoid nesting string functions because of this `LUA_BUFFERSIZE` issue in some firmware versions but you don't need to do a per character loop.  This is slow and per character concat is both slow and churns RAM.  So why not just do a pattern match looking for `%xx` like this:
``` lua
local function unescape(s)
  local rt, i, len = "", 1, #s
  s = s:gsub('+', ' ')
  local j, xx = s:match('()%%(%x%x)', i)
  while j do
    rt = rt .. s:sub(i, j-1) .. string.char(tonumber(xx,16))
    i = j+3
    j, xx = s:match('()%%(%x%x)', i)
  end
  return rt .. s:sub(i)
end
```
It's the same length as yours but this only does **n+1** loops where **n** is the number of escapes and **n+1** evaluations of `rt` so will run a lot faster.  In fact because of the function overhead of declaring a function this is actually faster and less of a memory footprint than the gsub callback technique.

Note that this does not attempt to process malformed escapes such as `%4g`.

TerryE, 2015-12-12:
See #719

closed 2015-12-12 by TerryE