nodemcu/nodemcu-firmware #314: tmr.time makes sudden jump

markkuhu, 2015-03-27:
tmr.times seems to make sudden jump when it reaches a value around 12000.

[image]
[image]
[image]
[image]

frankexoo, 2015-03-27:
This issue has been resolved already: use a later firmware version (20150318)

markkuhu, 2015-03-28:
Hi,
and thank you for the quick reply.

Two comments though.

I actually had updated the version - or so I thought. The control on
versions using the Nodemcu programmer is not that good. The version, or
better still the build date, could be made visible there.

The way I acquired the new version was to download the zip file and just
extract it to the same folder as the previous one. This was not a good idea.
Maybe something could be done to prevent the issues arising from my method
could be done?

________________________________
Markku Hukari
+ 358 (0)400 215 360

<https://github.com/nodemcu/nodemcu-firmware/issues/314#issuecomment-8710191
2> .
<https://github.com/notifications/beacon/ALJJZ2sKGulgsbAXTGITK8JYlubApkHWks5
n5cgjgaJpZM4D2BjS.gif>

devsaurus, 2015-12-13:
Issue was considered resolved.

closed 2015-12-13 by devsaurus