nodemcu/nodemcu-firmware #317: Update to eLua 5.2

hmic, 2015-03-29:
What about an update to a newer eLua Version?

simcop2387, 2015-08-24:
I'd love this as well but it's likely because eLua hasn't released a 5.2 version yet as near as I can tell.   In the master branch, it's still 5.1.4:

https://github.com/elua/elua/blob/master/src/lua/lua.h#L20

TerryE, 2015-08-24:
Guys, don't hold your breath on this one.  What added advantages does Lua 5.2 add for the embedded environment?  Will these justify the development effort?  

Given that nodeMCU is a fork of eLua and won't move ahead of eLua in terms of the Lua version, I suggest that we close this issue at least until there is a scheduled eLua version based on Lua 5.2.

closed 2015-08-24 by TerryE