nodemcu/nodemcu-firmware #318: LCD16x2 display.

asm7100, 2015-03-29:
Hi
Is it posebel to make somting to use a 16 x 2 LCD display, connectet to at MCP23017?

kubi57, 2015-03-30:
Hi,

yes, that can be done in lua. If you operate the LCD in 4-bit mode, an MCP23008 is also working.

asm7100, 2015-03-30:
Nice, has anyone don it in Lua code??

clixx-io, 2015-03-31:
I've done it in python. Shouldn't take much to convert to LUA.

https://github.com/clixx-io/clixx.io/blob/master/source/python/clixxIO_CharLCD.py

asm7100, 2015-03-31:
Sorry not that strong in python:-)

dvv, 2015-04-01:
https://github.com/dvv/nodemcu-thingies/blob/master/lcd1602.lua but it's for I2C

asm7100, 2015-04-02:
Hey dvv, MCP23017 is using i2c, its a 2 x 8 bit port extender, MCP23008 has only one.

devsaurus, 2015-12-13:
No WIP, thus closing as per #719.

closed 2015-12-13 by devsaurus