nodemcu/nodemcu-firmware #319: tmr.delay is too slow

klukonin, 2015-03-30:
It usually takes about 20-45 us to call tmr.delay() function.
So when I run tmr.delay(15) I get 45-80 us, or sometimes 180 us.
This really depends on cycle and counter direction.
For cycle 

for i =0, 16, 1 do
gpio.write(4, 0)
tmr.delay(15)
gpio.write(4, 1)
tmr.delay(15)
end

delay is about 180 us.

For cycle 

for i =16, 0, -1 do
gpio.write(4, 0)
tmr.delay(15)
gpio.write(4, 1)
tmr.delay(15)
end

delay is about 85 us.

In while cycle delay is about 45 us.

So is there any way to get a stable delay about 10-15 us ?

dvv, 2015-03-30:
First, report the same with functions `local`ized:
```lua
do
  local w, d = gpio.write, tmr.delay
  for i = 0, 16 do
    w(4, 0)
    d(15)
    w(4, 1)
    d(15)
  end
  ...
end
```

klukonin, 2015-03-30:
I did it.
Still too slow.
http://tinypic.com/r/14xoy2e/8

Also thre are some noise on GPIO0 and GPIO2
http://i59.tinypic.com/20t5kox.png

I don't know what is that.
Usually appear in normal boot.
But doesn't appear in firmware upgrade mode.
May be it should be another report for this.

nickandrew, 2016-02-10:
> So is there any way to get a stable delay about 10-15 us ?

Probably not from Lua. If you need more accuracy, write a driver in C. Closing.

closed 2016-02-10 by nickandrew