nodemcu/nodemcu-firmware #320: collectgarbage() seems to block system

stefan73, 2015-03-30:
Using the pre-build firmware 0.9.5 and a code quite similar to the example web-server webap_toggle_pin.lua

This does not run stable. The webserver is often unresponsive (no connection from browser) and becomes responsive again in-between. There is no restart happening as variables keep their values between unresponsive and responsive phases.

Commenting out the explicit call to the garbage collector at the end of webap_toggle_pin.lua seems to fix the problem. Monitoring node.heap() shows that garbage collector is active.

nodemcu, 2015-04-02:
do not need to use collectgarbage()
because the firmware do this automatically in a very aggressive way. 
it ALWAYS do GC when any chance available.

stefan73, 2015-04-02:
The sample webap_toggle_pin.lua in this project calls the garbage collector manually. And as people love "copy/paste programming" I guess that 2/3  of all nodemcu webserver code sniplets in the web meanwhile also call it.
If not required or even detrimental, I would recommend to take it out of the sample.

devsaurus, 2015-12-13:
Request to remove [collectgarbage() in webap_toggle_pin.lua](https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_examples/webap_toggle_pin.lua#L30).

closed 2016-01-07 by devsaurus