nodemcu/nodemcu-firmware #321: ws2812.writergb scrambles buffer

makefu, 2015-04-01:
Hi, i ran into the problem that the writergb function of the ws2812 module scrambles the buffer it uses by swapping first and second char for each rgb triple. I am using the buffer to hold the state of the LEDs. 

A simple re-assignment of the buffer to another variable does not seem to be enough to preserve the buffer.

Test:
```lua
> buf=string.char(1,2,3,4,5,6,7,8,9)
> buf2=buf
> =buf:byte(1,9)
1	2	3	4	5	6	7	8	9
> ws2812.writergb(7,buf)
> =buf:byte(1,9)
2	1	3	5	4	6	8	7	9
> =buf2:byte(1,9)
2	1	3	5	4	6	8	7	9
```

drv53b6, 2015-04-17:
yes, she do it. it not bug

makefu, 2015-04-17:
@drv53b6 i would also be fine with your suggestion, just keep grb , do not scramble the buffer. 
It just needs to be documented that it is not RGB or even provide ws2812.writergb() and ws2812.write() .

drv53b6, 2015-04-17:
@makefu make forck and request, rorrect both, and code and documentation

makefu, 2015-04-17:
@drv53b6 here at the stuttgart hackerspace in germany we call this `anhubeln` - trying to make someone else do what you think would be cool to have :D

When i find the time i will try to just revert to the original c code and have the additional c code for rgb still in there. of course it would be better if someone more experienced in writing code would do this haha

drv53b6, 2015-04-17:
@makefu i can but dont want

makefu, 2015-05-26:
@drv53b6 your anhubelei was super effective. 
please have a look at my pull request :)

PR #445 (2015-05-29): app/modules/ws2812.c: re-add ws2812.write(), fixes #321
Add the original ws2812.write() function which does not scramble the
buffer while preserving the 'new' ws2812.writergb.

With this everybody should be happy again.
This fixes #321 while staying backwards compatible
commit d9bb5855 (2015-05-29): Merge pull request #445 from makefu/dev

app/modules/ws2812.c: re-add ws2812.write(), fixes #321
closed 2015-06-17 by vowstar