nodemcu/nodemcu-firmware #322: MEM CHECK FAIL!!!

pastukhov, 2015-04-01:
I'm getting MEM CHECK FAIL!!! with 0.9.6-dev_20150331 on nodemcu devboard v0.9

klukonin, 2015-04-01:
Such problems can be a result of physical silicon fail.
Can you make a check on another module?

pastukhov, 2015-04-01:
Yes, look like something wrong with board.

closed 2015-04-01 by pastukhov

Mic-Key, 2015-04-01:
Hi,
may by related to
http://bbs.espressif.com/viewtopic.php?f=7&t=205&hilit=mem+check+fail&start=10#p881
?

pastukhov, 2015-04-01:
Thanks, Mic-Key!

crivano, 2016-04-04:
Hi, for me, the solution was to change de connection speed to 9600, after flashing. It is very strange, but connecting at 115200 wasn't working and was showing MEM TEST FAIL!!!. Switching CoolTerm to 9600 just fixes that. As far as I can tell, it makes no sense, but I tested many times and you should too! :)

marcelstoer, 2016-04-04:
9600 for `master` branch, 115200 for `dev` branch.

Sardashti, 2021-11-12:
See this video : https://www.youtube.com/watch?v=tMuC0FpZ-Zo