nodemcu/nodemcu-firmware #323: ds18b20 unknown condition....

gorec2005, 2015-04-01:
ds18B20 connected to gpio1(pin 10)...

if I try to read the raw data from ds is always returned one result (85')
> ow.setup(10);ow.select(10,a);ow.write(10,0x44,1);tmr.wdclr();ow.reset(10);ow.select(10,a);ow.write(10,0xBE,1);d=nil;for i=1,8 do print(ow.read(10)); end
80
5
75
70
127
255
12
16
> 
> 
and if i use ds18b20 module it read (85') too
> t=require("ds18b20");t.setup(10);print("Temperature: "..t.read().."'C");t=nil;ds18b20=nil;package.loaded["ds18b20"]=nil
Temperature: 85'C
but second measurement so good...
> t=require("ds18b20");t.setup(10);print("Temperature: "..t.read().."'C");t=nil;ds18b20=nil;package.loaded["ds18b20"]=nil
Temperature: 35'C
>

it's a bug or it's a feature? :-)

Mic-Key, 2015-04-01:
Hi,
insert 750ms delay (conversion time) between ow.write(10,0x44,1) and ow.reset(10).

gorec2005, 2015-04-02:
this did not help :-(

> ow.setup(10);ow.reset(10);a=ow.search(10);ow.select(10,a);ow.write(10,0x44,1);tmr.delay(1000000);ow.reset(10);ow.select(10,a);ow.write(10,0xBE,1);d=nil;for i=1,8 do print(ow.read(10)); end
80
5
75
70
127
255
12
16
> ow.setup(10);ow.reset(10);a=ow.search(10);ow.select(10,a);ow.write(10,0x44,1);tmr.delay(1000000);ow.reset(10);ow.select(10,a);ow.write(10,0xBE,1);d=nil;for i=1,8 do print(ow.read(10)); end
80
5
75
70
127
255
12
16
> ow.setup(10);ow.reset(10);a=ow.search(10);ow.select(10,a);ow.write(10,0x44,1);tmr.delay(1000000);ow.reset(10);ow.select(10,a);ow.write(10,0xBE,1);d=nil;for i=1,8 do print(ow.read(10)); end
80
5
75
70
127
255
12
16
> t=require("ds18b20");t.setup(10);print("Temperature: "..t.read().."'C");t=nil;ds18b20=nil;package.loaded["ds18b20"]=nil
Temperature: 85'C                                                                                                                                                                                                         
> t=require("ds18b20");t.setup(10);print("Temperature: "..t.read().."'C");t=nil;ds18b20=nil;package.loaded["ds18b20"]=nil                                                                                                 
Temperature: 34'C                                                                                                                                                                                                         
>

Mic-Key, 2015-04-02:
Hi,
my test code (also testing some table and string functions - I'm still lua newbie :-)
```
ds_pin = 3
ow.setup(ds_pin)

function printTemp()
  ow.reset_search(ds_pin)
  addr = ow.search(ds_pin)    -- 1st address
  while (addr ~= nil) do
    data = {}
    for i = 1, #addr do data[i] = string.format("%02X", addr:byte(i)) end 
    crc = string.format("%02X", ow.crc8(string.sub(addr, 1, 7)))
    print('\r\naddr:', #data, table.concat(data, ':'), crc)

    ow.reset(ds_pin)
    ow.select(ds_pin, addr)
    ow.write(ds_pin, 0x44, 1) -- start conversion
    tmr.delay(1000000)        -- wait 1s (>750ms)

    ow.reset(ds_pin)
    ow.select(ds_pin, addr)
    ow.write(ds_pin, 0xBE, 1) -- read scratchpad
    data = {}
    for i = 1, 9 do data[i] = ow.read(ds_pin) end

    strData = ''
    for i = 1, #data - 1 do strData = strData .. string.char(data[i]) end
    crc = ow.crc8(strData)
    print('data:', #data, table.concat(data, ','), crc)

    temp = data[1] + 256 * data[2]
    print('temp: ', temp / 16, '°C')
  
    addr = ow.search(ds_pin)  -- next address
  end
  print('--- no more addresses ---')
end

printTemp()
```
output:
```
addr:	8	28:BE:49:08:06:00:00:DD	DD
data:	9	164,1,75,70,127,255,12,16,218	218
temp: 	26.25	°C

addr:	8	28:31:FA:08:06:00:00:12	12
data:	9	154,1,75,70,127,255,6,16,120	120
temp: 	25.625	°C
--- no more addresses ---
```
HTH

MacDada, 2015-07-13:
I use [this version](https://github.com/nodemcu/nodemcu-firmware/blob/aaca524ce02fca3cccdeaf7a6ffd0c4640c84ecf/lua_modules/ds18b20/ds18b20.lua) and have the same problem – first measurement gives me 85C. A little workaround:

    local function measureTemp()
        local temp = ds.read(addrs[1], ds.C)
        print("Measured temp: "..temp)

        if (85 == temp) then 
            print("Measuring temp error, trying again...")
        else 
            return temp
        end
    end

vlasov01, 2015-09-21:
I have the same issue (NodeMCU 0.9.6). Thank you MacDada for the workaround!

> dofile("main.lua");
Total DS18B20, sensors: 0
85
Measured temp: 85
Measuring temp error, trying again...
Total DS18B20, sensors: 1
85
> dofile("main.lua");
Total DS18B20, sensors: 0
25.5
Measured temp: 25.5

TerryE, 2015-09-21:
If you are going to use a retry workaround then code it a little more efficiently:

```Lua
    local function measureTemp()
        local temp
 = ds.read(addrs[1], ds.C)
        print("Measured temp: "..temp)

        if (85 == temp) then 
            print("Measuring temp error, trying again...")
            return measureTemp()
        else 
            return temp
        end
    end

vlasov01, 2015-09-22:
Thanks TerryE. Yes, it is using less memory.

nickandrew, 2015-10-03:
Please close. This doesn't look like a firmware issue.

gorec2005, 2015-10-03:
sorry

closed 2015-10-03 by gorec2005